IFL-CAMP/iiwa_stack #98: [ERROR]configuration/configureSmartServo could not be called

MengChao1, 2018-01-18:
I want to change the control mode. so I try to roslaunch "/iiwa_tool_examples/launch/iiwa_tool_control_mode.launch" I get the error. I don't know why,  Thank you.

SalvoVirga, 2018-01-18:
Do you see the service `/iiwa/configuration/configureSmartServo` being advertised when running `rosservice list` ?
Those examples are very old, probably not valid anymore.

MengChao1, 2018-01-19:
Thanks for your answer. Now I have another question. Can I use the "/iiwa/configuration/configureSmartServo" service in gazebo?  When I "rosservice list" I can't find it?

SalvoVirga, 2018-01-25:
No sorry, that service is advertised directly by the robot controller, it will not exist in simulation. You would have to use the interfaces offered by Gazebo for that.

MengChao1, 2018-01-26:
Thanks for your reply

closed 2018-01-26 by SalvoVirga