IFL-CAMP/iiwa_stack #101: Use Windows symlinks instead of creating a new project containing ROSJavaLib and de

DavidWuthier, 2018-02-09:
Hi,

Windows actually supports symbolic links, and I'm using them for including the "ROSJavaLib" and "de" folders in my Sunrise Workbench project, works like a charm:

1. Open a command prompt as an administrator
2. cd \path\to\sunrise\project
3. mklink /D ROSJavaLib ..\relative\path\to\iiwa_stack\iiwa_ros_java\ROSJavaLib
4. cd src
5. mklink /D de ..\relative\path\to\iiwa_stack\iiwa_ros_java\src\de

Would you like to adopt this setup? Shall I update the wiki?

SalvoVirga, 2018-02-12:
Thanks a lot for the contribution!
I quickly changed the WIki description to fit your suggestion: https://github.com/IFL-CAMP/iiwa_stack/wiki/sunrise_project_setup

Feel free to change it the way you prefer if you think something is missing and close the issue afterwards :)

DavidWuthier, 2018-02-14:
You're very welcome!
Since Github doesn't support pull requests for wiki pages, you can merge my fixes as following

    git clone https://github.com/IFL-CAMP/iiwa_stack.wiki.git
    git remote add fork https://github.com/DavidWuthier/iiwa_stack.wiki.git
    git pull fork master
    git push

Thank you for the great work!

SalvoVirga, 2018-02-16:
Done! Thanks again!

closed 2018-02-16 by SalvoVirga