IFL-CAMP/iiwa_stack #102: Support for Work Visual 5 (5.0.2 or 5.0.4) for Sunrise OS 1.15?

maxpfingsthorn, 2018-02-16:
First of all, thanks for your efforts!

I was wondering if someone hat already tried the iiwa_stack on Sunrise OS 1.15 (released 1st Feb. 2018)? One large change is the upgrade of Windows to 3.7 and of Work Visual to version 5. A similar bump occurred with 1.11, which apparently broke backwards compatibility. That is why I am asking.

Or, in reverse, what is the latest version that was successfully tested with the iiwa_stack? 1.14.1?

SalvoVirga, 2018-02-16:
We still use 1.11 and I heard of people using 1.13 without problems.
I will poke the guys at KUKA to get 1.15, but given some upcoming deadline we will have time to test it (and hopefully close all the open features/issues) after mid March. PRs are always welcome!

husseinmn, 2018-02-28:
Hello, 
I wanted to know if there is people successfully using the iiwa_stack with sunrise 1.14? 
Thanks for the great work!

SalvoVirga, 2018-02-28:
@husseinmn Does 1.14 have a Connectivity (Servoing + FRI) module? Usually releases with an even number do not.

If it does, it should all be fine. If APIs from KUKA changed in that version, then few changes might be needed. I never tried that version myself.

husseinmn, 2018-02-28:
I don't have the robot yet, but in the quote from Kuka those modules are listed. I will try to find out if there are some big changes in the API.
Thanks for the quick reply!

SalvoVirga, 2018-02-28:
You should ask them to get the latest Sunrise OS + Connectivity version (which is 1.15), it is important that the Connectivity package (Servoing + FRI) is included.
From the last chat I had with the guys at KUKA, moving to 1.15 from a previous version is more annoying than usual, so be sure you have that version already when you get the robot.

I will update our robots to 1.15 in the next weeks.

kristoferB, 2018-04-18:
Hello @SalvoVirga. What is the status for 1.15? Did you have time to upgrade and try out the stack? Have you received any information about the changes for FRI and servoing from 1.11 from KUKA?

Thanks for all your great work!

SalvoVirga, 2018-04-22:
Small update:

Apparently, Sunrise 1.15 requires an updated version of the Windows Embedded on the Sunrise Cabinet. New robots will ship with the updated version and Sunrise 1.15, old robots need to be updated.
No update = No Sunrise 1.15

KUKA provides some recovery/update USB key (**at a cost**) to update the Cabinet, they were supposed to send me one weeks ago, I'll ping them again. Once I have that I can update one of our robots and check Sunrise 1.15.

kellerjo, 2019-05-17:
Hey there,
We want to update to 1.16.
Has anyone already tested this version?

exo-core, 2019-05-17:
We're on 1.16 and it's working fine :smiley:

ajardon, 2019-05-17:
We are working with Sunrise 1.16 and it's going fine¡¡¡

SalvoVirga, 2019-06-05:
Master now supports 1.15 and 1.16 after #178

closed 2019-06-05 by SalvoVirga