IFL-CAMP/iiwa_stack #104: How can I use force control with moving the arm together?

impet14, 2018-03-22:
Hi all,
I would like to control iiwa arm using force control to limit force when the robot hit something. 
How could I do that? I am the beginner for iiwa. 
Thanks guys

SalvoVirga, 2018-03-22:
It really depends on what you need to do. You could use the provided topics and listen to the current Joint Torques (`/iiwa/state/JointTorque`) and/or Cartesian Forces/Torques (`/iiwa/state/CartesianWrench`) and act accordingly when needed.

impet14, 2018-04-18:
Thanks for the reply. Can we control the arm for parallel force/position control?

SalvoVirga, 2018-04-22:
Again, it really depends what you need to do. But yes, you can control the robot in position control mode, switch to impedance mode and back, apply a fixed force while commanding positions etc.