IFL-CAMP/iiwa_stack #105: Use Helper Class iiwa_ros ERROR

dracotin, 2018-03-29:
I create a subscriber within a ROS Node to read robot joint position successfully,but when i use the helper class iiwa_ros, i can't get robot joint position. I just use the example that showed on the readrobotstate wiki, it get "No Joint Message available ". Also when i use class iiwa_ros to set joint velocity , it get error "configuration/pathParameters could not be called " , i check rostopic list ,the service is advertised.
Here is my code:
>#include <iiwa_ros.h>
int main(int argc, char **argv){
    iiwa_msgs::JointPosition joint_position;
    geometry_msgs::PoseStamped Cartesian_position;
    iiwa_ros::iiwaRos my_iiwa;
    //初始化node
    ros::init(argc,argv,"readstate");
    ros::NodeHandle nh("~");
    //my_iiwa初始化
    my_iiwa.init();
    bool success = true;
        if(my_iiwa.getJointPosition(joint_position)){
        ROS_INFO_STREAM("i got a Joint Message and its content is : "<<joint_position.position.a1<<" "<<joint_position.position.a2<<" "
        <<joint_position.position.a3<<" "<<joint_position.position.a4<<" "<<joint_position.position.a5<<+" "<<joint_position.position.a6<<
        " "<<joint_position.position.a7);
    }
    else{
        ROS_INFO_STREAM("No Joint Message available ");
    }
}

SalvoVirga, 2018-04-01:
How do you launch that executable? 

If you look at [iiwa_ros.cpp](https://github.com/IFL-CAMP/iiwa_stack/blob/master/iiwa_ros/src/iiwa_ros.cpp#L48) you can notice that subscribers/publishers are defined for topic names like `state/JointPosition` or `command/CartesianPose`.
The real topics are instead `/<robot_name>/state/JointPosition` and so on, so you need to launch the ros node from the right namespace, e.g. iiwa.

In a launch file you will have something like
```xml 
<node ns="iiwa" name="ANameForMyNode" pkg="PackageName" type="TargetExecutableName" />
```

or from console you can use

```shell
export ROS_NAMESPACE=iiwa
rosrun <package_name> <node_name>
```

Of course, if you set up a different name for your robot, change that accordingly.

dracotin, 2018-04-03:
I have a launch file like you said , but the console shows that 

> PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.13
NODES
  /iiwa/
    robotstate (iiwa_demo/readstate)
ROS_MASTER_URI=http://192.168.168.1:11311
process[iiwa/robotstate-1]: started with pid [24001]
[ INFO] [1522747131.931412440]: No Joint Message available 
[ INFO] [1522747131.931451353]: No Cartesian Pose Message available 
[iiwa/robotstate-1] process has finished cleanly
log file: /home/lab1811/.ros/log/536b68f0-3712-11e8-91c6-502b73e415b4/iiwa-robotstate-1*.log

it dosen't  work, `my_iiwa.getJointPosition(joint_position)`is false, the value of `joint_position` is zero . i wonder if my code is wrong , could you show me a cpmplete demo for read robot state .
thanks.

dracotin, 2018-05-02:
Sorry for my late feedback , i am new to ros , aslo C++. I read your iiwa stack examples , and i check my procedure and debug , finally i can read the robot state and command the robot . Now i try to use moveit to make motion planning for iiwa . Thank you a lot , this issue is solved .