IFL-CAMP/iiwa_stack #111: ROSSmartServo run failed

jacknlliu, 2018-04-22:
After starting ROSSmartServo, the control loop started, but some seconds later, it exits with

```
ROS control loop aborted. java.lang.IllegalStateException: Update of SPR did not succeed within 1 second(s). Its state is Active。
Stopping ROS nodes
Stopped ROS nodes
ROS control loop has ended. Application terminated.
```

SalvoVirga, 2018-04-22:
I just bumped the Sunrise 1.7 branch, please check the new version.
If possible, I always suggest to upgrade your Sunrise version (possibly to 1.13).

jacknlliu, 2018-04-23:
I add a line about sleeping 1ms to [ROSBaseApplication.java](https://github.com/IFL-CAMP/iiwa_stack/blob/master/iiwa_ros_java/src/de/tum/in/camp/kuka/ros/ROSBaseApplication.java#L222), and it solved this issue.

```
import com.kuka.common.ThreadUtil;

			while(running) { 
				decimationCounter++;

				// This will publish the current robot state on the various ROS topics.
				publisher.publishCurrentState(robot, motion, toolFrame);
                               
                              // sleep 1ms for avoiding "Update of SPR did not succeed within 1 second"
                               ThreadUtil.milliSleep(1);

				if ((decimationCounter % controlDecimation) == 0)
					controlLoop();  // Perform control loop specified by subclass
			} 
```

SalvoVirga, 2018-04-23:
To me this sounds a bit random, but if it does the job for your case I guess is ok.
Try anyhow to update to a more recent Sunrise version if you can.

closed 2018-04-23 by SalvoVirga