IFL-CAMP/iiwa_stack #113: Subscriber in java side not receiving all the commanded target points

TheCamusean, 2018-05-16:
Hi,

I added a counter in the callback of /command/JointPosition in the java side.

I created a trajectory of 569 points and sent it in a frame rate of 30Hz.

My subscriber is only receiving 100 points, so is losing a fifth part of it. I consider 30Hz to be slow enough frequency rate, but I am not sure. 

Any Suggestions?

SalvoVirga, 2018-06-25:
That is very strange, I'll try to look into this in the next days.

TheCamusean, 2018-07-04:
I tried hard on this and I got a maximun communication rate of 10Hz. Seems like subscriber needed lot of time to deal with the data. I was little bit surprised also. I was expecting something much more faster. This was affecting me at the end in strange behaviours coming from the SmartServo.

It was really poor so I jumped to TCP/IP communication getting a communication of 120Hz.

lihx108, 2019-03-14:
> I tried hard on this and I got a maximun communication rate of 10Hz. Seems like subscriber needed lot of time to deal with the data. I was little bit surprised also. I was expecting something much more faster. This was affecting me at the end in strange behaviours coming from the SmartServo.
> 
> It was really poor so I jumped to TCP/IP communication getting a communication of 120Hz.
> 
> Moreover, I also had the opportunity to work with a robot from another group(CNR-ITIA) and they were facing similar problems.

I meet the same problem, and the sunrise java side receive topic at a frequency of 5HZ. How did you solve it?

JimmyDaSilva, 2020-02-24:
I have been having delay issues as well, leading to jerky motions as described in the Troubleshooting section of the `Sunrise.Servoing 1.15` manual.

For instance, here are the responses to steps and triangles for the last axis, for a command loop at 500Hz using the PositionControl mode.
[image]
[image]

@SalvoVirga, have you ever had issues like this in the past ?
Is there a way to properly play around with the `setMinimumTrajectoryExecutionTime(…)` and `setTimeoutAfterGoalReach` functions ?
Also, what control frequency people usually use when using iiwa_stack or the SmartServo class in general ?

Basically I also get the jerky motions when I use the MoveIt package.
Maybe there is something wrong with my network communication...
I even turn on the use of the NTP server, but that didn't solved the problem.

Many thanks,
Jimmy