IFL-CAMP/iiwa_stack #114: master-Sunrise1.7 ROSSmartServo trigged robot brakes closed after a few minutes

jacknlliu, 2018-05-18:
Starting ROSSmartServo a few minutes later (3-4 min), the robot brakes closed automatically. The robot can't work longer like this. But If we switch the robot to different control mode, and switch it back, the brakes will open again. But a few minutes later (3-4 min), brakes closed again!

We use `master-Sunrise1.7` latest commit [9a0350f](https://github.com/IFL-CAMP/iiwa_stack/commit/9a0350f8833821b4993627c5086158fbe186f7b4). But the previous commit [f65affe](https://github.com/IFL-CAMP/iiwa_stack/commit/f65affe9a8cde92bf266cb06ff0e70539626ee08) before commit [Merge branch 'master' into master-Sunrise1.7](https://github.com/IFL-CAMP/iiwa_stack/commit/0d3ab869f26a3f3b171e0341182ee85335a74ac0) doesn't meet this issue.

SalvoVirga, 2018-06-25:
Hum, this should be the value set by [setTimeoutAfterGoalReach](https://github.com/IFL-CAMP/iiwa_stack/blob/master/iiwa_ros_java/src/de/tum/in/camp/kuka/ros/ROSSmartServo.java#L341). It represent the number of seconds to wait for another command before triggering the breaks. You can change that to whatever fits best your application.

In general, however, I'd suggest you to update your robot to the latest supported/tested version of Sunrise.

SalvoVirga, 2019-06-05:
Dropped native support for Sunrise 1.7 for the moment, a custom version of the new release of iiwa_stack could be set up for it upon request tho

closed 2019-06-05 by SalvoVirga