IFL-CAMP/iiwa_stack #115: iiwa_ros_java/src not working with Sunrise workbench (wiki instructions)

emielke12, 2018-06-14:
Hi, I am following along with the wiki instructions, and I got to the sunrise workbench portion of the instructions [https://github.com/IFL-CAMP/iiwa_stack/wiki/sunrise_project_setup](url) and I my sunrise workbench has red Xs for all the files from iiwa_ros_java/src. I've attached some screenshots. It looks like there are classes from a com.kuka.connectivity file that are missing. Additionally, I didn't see the same options for the StationSetup.cat software tab. Are these related? Is there a problem with the smartServo class? Is there a way to get the other software to show up? 
[image]
[image]
[image]

SalvoVirga, 2018-06-14:
Yes, the problems are related!
You need the Sunrise.Connectivity (Sunrise.Servoing) to use iiwa_stack (or in general to control the robot at runtime). This is an additional package that you have to request from KUKA and install into the Sunrise Workbench environment.

closed 2018-06-19 by emielke12

madhukartlr, 2019-06-27:
@SalvoVirga I have already installed the Sunrise.Connectivity using **com.kuka.connectivity.industrial.repository-1.7.0.ConnV17_2** but I am still having issues with importing Servo Motion controls. 
Thanks!