IFL-CAMP/iiwa_stack #116: Error while adding a tool on the end-effector

wurui0213, 2018-06-19:
hello everyone~
I have already calibration the tool and synchronize the project, safetyconfiguration also been changed. 
when i use the impedance mode in application "FeatureDemo", every thing is OK and the gracity of the tool is compensated well.
But when i run "ROSSmartServo",it shows error"No tool attached. using flange" and "Load Mass validation failed axisTauExtMsr[1] of .... is 14.80 should be less than 10", while actually the tool calibration in the robot is just 2.6Kg, and i can't find out why this happen.
Thank a lot for help.

SalvoVirga, 2018-06-25:
Hi!

iiwa_stack uses a ROS Parameter to let you select the tool to use within the Sunrise application, i.e. the tool that you attached to the robot and defined/calibrated in Sunrise.
You could just set the right ROS Parameter and when you start ROSSMartServo the tool with the given name is selected. I suggest anyhow to also create an URDF description to load the robot model that also contains the tool you want to you use.

More or less everything is explained in this [wiki page](https://github.com/IFL-CAMP/iiwa_stack/wiki/attachtool). Let me know if something is unclear.

wurui0213, 2018-06-25:
Thank you for your reply, we will try this wiki and report our attempt in future~

hitli, 2018-07-05:
We tried as wiki said but still failed.
We created an URDF description and set the ROS Parameter.But when we run "ROSSmartServo", it shows "Attaching tool Toolone" and error "Load Mass validation failed axisTauExtMsr[1] of .... is 14.80 should be less than 10" as before.If we use a light tool and run "ROSSmartServo", it shows "Stopping task due to the following error: java.lang.NullPointerException at de.tum.in.camp.kuka.ros.ROSBaseApplicarion.run(ROSBaseApplicarion.java:199)".By the way, we are using branch for sunrise 1.9.