IFL-CAMP/iiwa_stack #119: Edit Master IP via Pendant

emielke12, 2018-07-02:
Is there a way to change the master IP without having to change the config.txt file and reinstalling the sunrise package? Is there a way call up a dialog on the pendant to change the master on the fly?

mdicicco, 2018-07-02:
Just put an entry in your RoboticsAPI.data.xml file.  You can change the value in the pendant's "Process Data" tab and then run your program.  In your code you access it with getApplicationData().getProcessData(<your item name>).getValue()

The only issue is that it will reset back to the default after a re-deploy of your code.  How often do you change your ros master IP address?  I've seen some posts about using dchp to dynamically find the IP address via the hostname.  That's probably a better solution if you find it changing a lot.

emielke12, 2018-07-02:
So the issue is more that there are multiple users, and no dedicated computer for the iiwa (for the time being.) So it would be nice to have a way to change it as needed, and your first solution sounds like it will work for now. 

I am not super clear on what I should be putting in the entry for the RoboticsAPI.data.xml file (i.e how to format the entry.) Do you have an tips for that or where I could look for an example?

mdicicco, 2018-07-02:
My iiwa came with an example file called RoboticsAPI.data.ProcessDataExample.xml that had example xml for each of the different types (float, int, string, etc) that you can define for your variables.  All the variables need to be inside of a processDataContainer tag.  I'll try to put a snippet below:
```
<processDataContainer>
      <processData displayName="Example Integer" dataType="java.lang.Integer" defaultValue="0" id="counter" min="0" max="100" value="82"/>
      <processData displayName="Example Double" dataType="java.lang.Double" defaultValue="13.37" id="itemCount" min="0.01" max="200.7" value="12" unit="Objects" comment="already produced"/>
      <processData displayName="Example String" dataType="java.lang.String" defaultValue="0" id="someName" value="HelloWorld"/>
      <processData displayName="Example Boolean" dataType="java.lang.Boolean" defaultValue="false" id="isTrue"  value="true"/>
      <processData displayName="Example InvisibleOnHmi" dataType="java.lang.Boolean" defaultValue="false" id="invisibleOnHmi" value="true" visibleOnHmi="false"/>
      <processData displayName="Example NotEditableOnHmi" dataType="java.lang.Boolean" defaultValue="false" id="notEditableOnHmi" value="true" editableOnHmi="false"/>
</processDataContainer>
```

SalvoVirga, 2019-06-08:
This has been solved with #178 !

closed 2019-06-08 by SalvoVirga