IFL-CAMP/iiwa_stack #130: torque overlay/control

gpldecha, 2018-10-16:
I am interested in command the robot by sending external torques. Does the stack support this functionality ? From my understand it would require to be in joint impedance ctrl mode, set the stiffness to zero and overlay the external joint torques (similar to the lwr4). This would be very similar to the LBRTorqueSineOverlay example from the KUKA FRI API examples. 

I am currently trying to do this with the FRI but the SmartServo (this stack) approach could also be an option. I do not yet fully understand the pro/cons of SmartServo vs FRI. I get that the smart servo is running ros on the cabinet, but apart from that I am unsure. 

Any incites regarding these two questions, would be greatly appreciated.

ajardon, 2019-05-09:
I would like to try something similar. Did you get any result?

gpldecha, 2019-05-10:
@ajardon You cannot send torques with this package. You would need to implement your own java FRI interface. I recommend you checking out [robot-forum](https://www.robot-forum.com/robotforum/cobot-kuka/)

ajardon, 2019-05-10:
thanks ¡¡¡ I'll do it.