IFL-CAMP/iiwa_stack #131: How to change velocity during executing trajectory?

michtesar, 2018-10-29:
Hello, I would urgently need in my application to change the speed of the robot based on external measurements.

Is there any way to plan a trajectory and while executing it, the robot could change a speed?

Thank you.

michtesar, 2018-10-31:
Hi exo-core,

that is pity, but great you solve it this way. I was thinking about perceptual changing the simulated time. I was playing with `/iiwa/configuration/setPathParameters` but with no luck as you say. Are you able to share the code (at least the part with `JointVelocityController`?

Thank you very much,

Michael

SalvoVirga, 2018-10-31:
If you are brave enough, you could experiment a bit with the development branch and control the robot directly using Cartesian velocity, but that branch is still not too stable.

exo-core, 2018-10-31:
@michtesar I asked my colleague who did the changes to look for the code :wink: 

@SalvoVirga I spend some time experimenting with the development branch as we also needed to control the Cartesian speed for safety certification of our use-case. I managed to port TF to rosjava and cross-compile it to run on the Sunrise Cabinet, to correctly interpret the PoseStamped messages. I also build a ROS action interface for those kind of motions and they generally work fine now :smiley: You can find the changes in [this repository](https://github.com/exo-core/iiwa_stack/tree/master).

SalvoVirga, 2018-10-31:
@exo-core not bad! I should finally find some time to work again on iiwa_stack soon. I also saw the PR, I will try to integrate everything nicely, maybe I'll ask you more info or to rework something (if ok for you :D)

exo-core, 2018-10-31:
@SalvoVirga Good to hear that. I was afraid the project was somewhat getting abandoned... Regarding the PR: It might be a bit outdated by now... If I remember correctly I created it after I managed to run TF on the Cabinet so it misses the whole ActionLib intergration, as I did this on a separate branch...

By the way, you should also update the Wiki on how to compile messages for Java 1.6. I was experimenting with VMs and old ROS versions for quite some time, but in the end I managed to cross compile everything on Ubuntu 16.04 / Kinetic following [this instructions](https://github.com/rosjava/rosjava_core/issues/253).

exo-core, 2018-11-16:
> Hast du den Code noch?
>
>
>
> -------- Forwarded Message --------
> executing trajectory? (#131)

Hi,

I uploaded the code under https://github.com/Mike2208/variable-joint-velocity-controller. The main changes are a service called 'joint_velocity_service' which can be used to modify the execution
velocity, as well as a change in velocity calculation.

For the velocity changes I added an extra variable to the TimeData struct called velocity_uptime. Basically, it simulates a slower execution time, which makes the robot move slower. 

Sorry for the delay in replying, I hope it helps. If you have any questions, don't hesitate to ask.

Regards,
Michael

michtesar, 2018-11-16:
Arne thank you and say thank Michael too :+1:

closed 2019-08-01 by michtesar