IFL-CAMP/iiwa_stack #132: command/CartesianPose not being Published

Andreluizfc, 2018-10-30:
Hi,

first of all I would like to leave my sincere gratitude for the effort in making this repository available to the ROS community.

I have a problem running the example ["using tools"](https://github.com/IFL-CAMP/iiwa_stack/wiki#using-tools) in Simulation. Compilation is ok, so are the following commands:

```
roscore
roslaunch iiwa_gazebo iiwa_gazebo.launch model:=iiwa7
roslaunch iiwa_tool_examples iiwa_tool_command.launch
```

Rviz starts and the application gets stuck at the line 52 of [this](https://github.com/SalvoVirga/iiwa_stack_examples/blob/master/iiwa_tool/iiwa_tool_examples/src/command_robot.cpp) cpp code, since there's nothing publishing on the topic `/iiwa/command/CartesianPose`.

Is there any other **launchfile** I am supposed to run?

Appreciate any effort dedicated at this matter. Thanks in advance,

André Castro.

____
Debug Info:

```
~$ rostopic echo /iiwa/state/CartesianPose 
WARNING: no messages received and simulated time is active.
Is /clock being published?
```
```
~$ rostopic echo /iiwa/command/CartesianPose 
WARNING: no messages received and simulated time is active.
Is /clock being published?
```

SalvoVirga, 2018-10-31:
Oh well, those examples are not really meant to run with the simulation environment :|
You can take some inspiration from them maybe, but you cannot use them as they are with Gazebo (they are also very old, so maybe you should not use them in general).

In simulation you won't have any of the topics/services that are provided by the real robot, you should rather use directly ros_control/MoveIt!

Andreluizfc, 2018-10-31:
Thank you for fast reply, @SalvoVirga. Since we are having connection problems with the real robot, I've been trying to test the in simulation.  Cheers!

closed 2018-10-31 by Andreluizfc