IFL-CAMP/iiwa_stack #134: Trying to view states and control them using a ROS program

avpai, 2018-11-18:
Hi,

I am a university student and very new to ROS and its programming. I am trying to control the position of the end effector of the KUKA arm usnign keyboard commands or some flag variables. I am thinking of doing the simualtion first and then using the code to control the actual KUKA robot. I have succesfully installed your library. I was able to set up a Gazebo simulation and the MoveIt tool using Rviz. Here are the results:

file:///home/tian/Pictures/gazebo.png

Then I followed your instructions on how to attach a tool and create a moveit package. I got that done and was able to view the Rviz simulation. I was able to control the position of the end effector using the mouse. 

file:///home/tian/Pictures/launch.png

file:///home/tian/Pictures/rviz.png

Now, as I wanted to control the end effector using a ROS program ( using key strokes/ some customizable variables), I looked into the list of rostopics that are being published:

file:///home/tian/Pictures/list.png

I tried to read the states using rostopic, but I could'nt. I kept getting the following errors:

file:///home/tian/Pictures/msg1.png

file:///home/tian/Pictures/msg2.png

I don't understand how to set a clock in the simulation to get the topics published. If I could get the cartesian position of the end effector published and get it to move using my keyboard, I would really be very happy. But since I a very new to ROS, I would really appreciate it if I could get some help/ pointers from people who have already done this.  Thank you

emielke12, 2019-02-12:
For future reference, you should use Markdown to submit code and errors, rather than pictures. 

The error `Are your messages built` indicates to me that you have not sourced your workspace properly. If you have sourced the iiwa_msgs workspace, you should not see the `Cannot load message class` error anymore. 

As far as the `rostopic echo /iiwa/state/CartesianPose`, you will not be able to read these states unless you are connected to a real-world robot, i.e. using `sim:=false` in the launch command. Unfortunately, at the current state of this repository, there are no `iiwa/state/` topics being published in simulation, and publishing to `/iiwa/command/` topics will not do anything in simulation. You'll need to go through the MoveIt framework if you want to do things in simulation.

avpai, 2019-02-13:
Hi,
Thanks. I found out that the gazebo library given here doesn't publish anything for the ROS to read. Now I am working on the KUKA arm itself rather than a simulation :)

closed 2019-02-13 by avpai