IFL-CAMP/iiwa_stack #138: sunrise 1.15.I have install servoing1.15.but The import com.kuka.connectivity.motionModel cannot be resolved

lihx108, 2018-12-28:
The import com.kuka.connectivity.motionModel cannot be resolved

lihx108, 2018-12-28:
[image]

lihx108, 2018-12-28:
oo..I made a mistake.sorry

closed 2018-12-28 by lihx108

madhukartlr, 2019-06-27:
I am getting the same issue. Would you mind telling me how it can resolved? Thanks.

lihx108, 2019-06-27:
> I am getting the same issue. Would you mind telling me how it can resolved? Thanks.

Do you install servoing1.15 by the correct steps? In [help]->[Install new software]->[add].......
[image]

madhukartlr, 2019-06-27:
Is it already inside the Sunrise workbench or do I need to ask KUKA for Sunrise.Servoing?

SalvoVirga, 2019-06-27:
It doesn't come with a standard installation of Sunrise Workbench, you have to install the Servoing package additionally.