IFL-CAMP/iiwa_stack #141: Development branch won't connect to `melodic` master

matthiashh, 2019-01-23:
_First of all thanks a lot for the time and effort you spent to write this software. I really appreciate not to have to start from scratch._

I am currently trying to set up two iiwa 7kg arms running SunriseOS 1.15 in combination with a Ubuntu 18.04 running ROS melodic. Because of #139 I tried to go with the `development` branch on both the Linux machine and the robots.

It turned out that the robot would not reliably connect to my `melodic` master. For some reasons it managed to do that once, but the other times it would be waiting without anything happening.
I also tried to connect the `development` Sunrise project to a `kinetic` master and that worked seamlessly. 

As a workaround, I equipped another Sunrise project with the code from the `master` branch and installed that on the robot. That one connected to my `melodic` master right away.

I haven't done any further investigations yet, but might be able to do some after I got myself a hold on the software stack and how to use it. But of course, if you have any specific recommendations or questions I am happy to contribute.

exo-core, 2019-01-24:
Hi @matthiashh ,

I assumed you checked for all the "standard problems", including:

- Wrong master address set in Sunrise project
- `ROS_IP` and `ROS_MASTER_URI` not configured or wrongly configured on PC running ROS-Master
- Restarting both, PC running roscore and the Sunrise Cabinet

Also, did you ran the `kinetic` and the `melodic` masters on the same machine? I noticed the Sunrise Controller can be quite picky on the network interfaces that it talks to (e.g. we have a Dell docking station in our lab and no mater what you do, you cannot connect to the Sunrise Cabinet when using it's network interface).

If the problem still exists: the used version of `rosjava` has been updated from `0.1.6` on `master` to `0.3.4` on `development`. You could try to replace the `*.jar` with the old version and see if this helps...

matthiashh, 2019-01-24:
I'll number your bullet points to answer them.

1. Configuration in the Sunrise project: There is one 'issue' hidden here, that I didn't mention: I placed the `config.txt` file in the same spot it should be in the `master` branch, but it wouldn't read it. So as a workaround I set `ROS_IP` and the `ROS_MASTER_URI` in the Java source code. It looked correct in the debug output on the smartpad + it connected to the `kinetic` master when just exchanging the IP address.
2. I can connect to the `melodic` master from another machine running `melodic` as well as one running `kinetic`
3. I restarted the Sunrise cabinet as well as the ROS master multiple times, but did not restart the machine running the master since the error pattern didn't look like that would help.

The `kinetic` and `melodic` master ran on different machines, but as said before I could connect to my `melodic` master from other machines as well as with the `iiwa_stack` Java part when installing it from the `master` branch.

I can try to take the old `rosjava` though and see if it helps.

SalvoVirga, 2019-01-24:
Hi, 

mind that there are plenty of changes implemented in the `development` branch with respect to `master`. Unfortunately these are still not well documented and tested. So, although, everything should already work nicely, consider `development` as non stable.

In particular to your problems:
- in development the **config.txt** file is not used anymore. All those configurations are now handled by some `Process Data` fields in Sunrise, unfortunately I just realized that all that is contained in your Sunrise project file `RoboticsAPI.data.xml`, which is not part of iiwa_stack. So here is what that file should contain:

```xml
<processDataContainer>
        <processData dataType="java.lang.String" defaultValue="iiwa" displayName="Robot Name" editableOnHmi="true" id="robot_name" value="iiwa"/>
        <processData dataType="java.lang.String" defaultValue="172.31.1.150" displayName="ROS Master IP" editableOnHmi="true" id="master_ip" value="172.31.1.150"/>
        <processData dataType="java.lang.String" defaultValue="11311" displayName="ROS Master Port" editableOnHmi="false" id="master_port" value="11311"/>
        <processData dataType="java.lang.String" defaultValue="172.31.1.147" displayName="Robot IP" editableOnHmi="false" id="robot_ip" value="172.31.1.147" visibleOnHmi="false"/>
        <processData dataType="java.lang.Boolean" defaultValue="false" displayName="Enable NTP" editableOnHmi="true" id="ntp" value="false"/>
</processDataContainer>
```

Once you add that to the file and you sync the project, you can access all these variables and modify them from the `Process Data` pane within the SmartPad. That include the IP of the ROS Master.
- only the **X66** ethernet port is needed in development, the same port you use to synchronize the project is now used for ROS.

So, connect to the X66 and be sure that you can ping the robot from the machine there runs the ROS Master (`ping 172.31.1.147` if you didn't change the robot default IP). Check on which interface your ROS Master will run, that is defined in `~/.bashrc`, be sure that is using the right subnet (172.31.1.X).
From the Process Data pane, be sure that the `ROS Master IP` is set to the right one, if not, you can modify it directly from there.
Start `roscore` and `ROSSmartServo`, everything should be good 👍 

I hope this can help, but I wrote it quickly thinking of all the steps you might need. Let me know if it helps or if you something is still wrong.

matthiashh, 2019-01-24:
Thanks a lot. I'll look into it. 
But now that I have a running setup, I first want to get a hold on how to use the arms with MoveIt. 
So probably a few days will pass, depending on how things and my schedule are evolving.

matthiashh, 2019-03-20:
I gave it another try and got this running it the first time:

[image]

Running it the second time it didn't crash though.

matthiashh, 2019-04-03:
I got it to work with the `development` branch for me.

The first problem was that I screwed up changing one of the IP addresses in `RoboticsAPI.data.xml`. After that I could receive the state of the robot, but not command it.

In order to be able to command it I needed to define my master with `ROS_IP` rather than `ROS_HOSTNAME`. Strictly speaking the [roscore setup](https://github.com/IFL-CAMP/iiwa_stack/wiki/roscore_setup) page says so, but it doesn't mention that `ROS_HOSTNAME` wouldn't work.

closed 2019-04-03 by matthiashh