IFL-CAMP/iiwa_stack #152: the .getRobotisConnected() is never true

avpai, 2019-02-25:
Hi,
I am new to ROS and I am using the KUKA arm for a project. The idea of the project is to move the end effector using inputs from keyboard/LAN/set variable,etc. So I have written a code:

```     
#include "ros/ros.h"
#include "std_msgs/String.h"
#include <iiwa_ros.h>
#include <conversions.h>
 
iiwa_msgs::JointPosition my_joint_position;
iiwa_msgs::JointPosition my_joint_position1;// =  iiwa_ros::jointQuantityFromDouble(0.0);

iiwa_ros::iiwaRos my_iiwa_ros_object;

void getdata(const iiwa_msgs::JointPosition::ConstPtr& msg)
 {        
       my_joint_position = *msg;	
      // ROS_INFO_STREAM("I got a joint Message and its contents is :" << my_joint_position.position.a6);	
   }

int main (int argc,char **argv)
 {
     ros::init(argc,argv,"iiwa_ros1");
     ros::NodeHandle n;

     ROS_INFO("Started");
	
   // sets up ros spin check
     ros::AsyncSpinner spinner(2);
     spinner.start();
     my_iiwa_ros_object.init(); // initialize iiwa_ros class

     //ros::Subscriber sub = n.subscribe("iiwa/state/JointPosition",1000,getdata);

      my_joint_position1.position = iiwa_ros::jointQuantityFromDouble(0.0); // create joint positions with all zeros

       my_iiwa_ros_object.setJointPosition(my_joint_position1); // This should set JointPosition
	ROS_INFO("Msg Sent");

	ROS_INFO("Spinning");
	//ros::spin();

	ros::Rate rate(30); // pauses program 10 times a seconds
	
	while(true)
	{	
		ROS_INFO_STREAM("Checking.." << my_iiwa_ros_object.getRobotIsConnected());
		if(my_iiwa_ros_object.getRobotIsConnected()) // This should be true while robot is connected
		{
			ROS_INFO("Connected");
		}
	
		rate.sleep(); // Pauses to keep running at 10hz
	}
	

	return 0;
}
```

The idea is to subscribe to the joint position of the end effector and then publish new co-ordinates to the end effector. I am getting the position of the end effector by subscribing to it ( I commented those section off in the program above), but I am not able to publish new co-ordinates into the arm. The program always skips the 
```
if(my_iiwa_ros_object.getRobotIsConnected())
```
condition. I am trying to figure out why this is happening . Any ideas or suggestions will be really appreciated. Thank you :)

marcoesposito1988, 2019-02-25:
Hi avpai,

Have you checked that the robot is looking for the master at the correct IP address? Can you see the robot nodes in `rosnode list`?

You can also try to echo the robot joint position with `rostopic echo /iiwa/jointStatePosition` (or something similar, I am writing from the phone.. you should see the correct topic name in `rostopic list`.

If you can’t see the nodes and/or the topics, or you can’t echo the topic, you might have to fix the iiwa config file in the sunrise project with your workstation ip.

Please check also that you have set the ROS_IP env variable correctly, as stated in the docs.

Marco Esposito
Research Assistant

Computer Aided Medical Procedures (CAMP)
Technische Universität München
Institut fuer Informatik I16
Boltzmannstr. 3<x-apple-data-detectors://4/0>
85748 Garching<x-apple-data-detectors://4/0> b. Muenchen, Germany
campar.in.tum.de<http://campar.in.tum.de/>

E-Mail: marco.esposito@tum.de<mailto:marco.esposito@tum.de>

avpai, 2019-02-26:
Hi Mark,
I am able to see the joint state using echo. I am also able to subscribe to joint position and I can see it being printed out in the terminal. The problem is in trying to publish the new co-ordinates into the kuka arm.

marcoesposito1988, 2019-02-28:
That looks fine. 

If I remember correctly, the robotIsConnected waits to receive the first message. So the problem should already be about receiving messages. If you can receive them with `rostopic echo`, that should mean that the problem is your node. Maybe it is running in the wrong namespace? I think it should be in the same namespace of the robot; if the robot is `/iiwa/iiwa_*`, then your node should be `/iiwa/mynode`.

However, I haven't used the robot in one year now, maybe @SalvoVirga can be more helpful?

SalvoVirga, 2019-03-04:
While running that code, check the output of rostopic info `/iiwa/state/JointPosition` if your node is among the subscribers to that topic and you can receive messages from the robot, then `getRobotIsConnected()`.
Try also to get the current joint position directly from `my_iiwa_ros_object` instead of creating another subscriber.

avpai, 2019-03-04:
Changed the main function to this one (below)
```
int main (int argc,char **argv)
{
	ros::init(argc,argv,"iiwa_ros");
	ros::NodeHandle n;
	ROS_INFO("Started");
	my_iiwa.init(); // initialize iiwa_ros class   	
        ros::Subscriber sub = n.subscribe("iiwa/state/JointPosition",1000,getjoint); 
        ros::Rate loop_rate(0.2);

  int count = 0;
  while (ros::ok())
  {
   // std_msgs::String msg;
    my_joint_position.position = iiwa_ros::jointQuantityFromDouble(0.17,1.49,2.58,0.63,0.63,-1.32,1.51); // create joint positions

    ROS_INFO("publish");  
   my_iiwa.setJointPosition(my_joint_position);  
    ros::spinOnce();
    loop_rate.sleep();     

  }
	return 0;
}
```
This seems to do the trick. I am now able to use the setJointPosition() function to control the joints of the kuka arm. I also added genmsg to the find_package section of the CMakeLists.txt. I will try using the getJointPosition function so that I can avoid using another subscriber.

SalvoVirga, 2019-03-04:
It does the job cause you are not checking anymore `getRobotIsConnected()` :P
Since you find your solution I close the issue, please reopen it you somehow have still problems with `getRobotIsConnected()`, which should work finely.

closed 2019-03-04 by SalvoVirga