IFL-CAMP/iiwa_stack #153: Using actionlib in python [development branch]

gmanwal, 2019-02-28:
I am currently attempting to use the actions that have been developed in the development branch (iiwa/action/move_to_cartesian_pose etc.) in order to get around the problems with moveit linear motion paths.

I am writing the actionclient in python (something like this):
<code>
position_2 = Pose(position=[-0.30744, 0.1505, 0.6], orientation=[0.976476550102
,-0.1154, -0.1813, -0.0166])
    pose_2 = CartesianPose(poseStamped=PoseStamped(pose=position_2, header=header))

 # Creates the SimpleActionClient, passing the type of the action
    # (FibonacciAction) to the constructor.
    client = actionlib.SimpleActionClient('iiwa/action/move_to_cartesian_pose', MoveToCartesianPoseAction)

    # Waits until the action server has started up and started
    # listening for goals
    client.wait_for_server()
<code>

The client seems to hang when waiting for server.

I thought it might have been a namespace problem at first so I changed "iiwa/action/move_to_cartesian_pose" to "move_to_cartesian_pose" in java actionServer file in Sunrise.

I was wondering if anyone else has tested this and had any luck?

George

chenhaowen01, 2019-03-04:
I don't test this action but '/iiwa/action/move_to_cartestian_pose_lin' action. And it work on my side.

SalvoVirga, 2019-03-04:
> I thought it might have been a namespace problem at first so I changed "iiwa/action/move_to_cartesian_pose" to "move_to_cartesian_pose" in java actionServer file in Sunrise.

Does this fix the problem?