IFL-CAMP/iiwa_stack #157: ERROR: Cannot load message class for [iiwa_msgs/JointPosition]. Are your messages built?

marcoojer, 2019-03-13:
Hi! I've just connect my iiwa 14 (sunriseOS 1.13) with ROS kinetic in Ubuntu 16.04, but when I  try to read the joinsState via ```rostopic echo /iiwa/state/JointPosition``` I get the following error: ERROR: ```Cannot load message class for [iiwa_msgs/JointPosition]. Are your messages built?```
This behavior is very strange since I'm able to read the topic ```/iiwa/state/CartesianPose```
Does anyone have an idea about how to solve that?? Thanks in advance

SalvoVirga, 2019-03-13:
You can receive messages from `/iiwa/state/CartesianPose` because they are `geometry_msgs::PoseStamped`, which are defined in the global ROS installation, you need to source the workspace you used to build `iiwa_stack` since the `iiwa_msgs::JointPosition` message is defined there.

i.e. run `source devel/setup.bash` from the root of the workspace.

You need to do that for every console instance or you can add `source <PATH_TO_WORKSPACE>/devel/setup.bash` to your `~/.bashrc` file, just like you did during the [ROS installation for /opt/ros/kinetic/setup.bash](http://wiki.ros.org/kinetic/Installation/Ubuntu).

marcoojer, 2019-03-14:
I already tried that, but it does not work. I can see the ```iiwa_msgs``` with ```rosmsg list``` and the info of every message with ```rosmsg info```, but when echoing the topic the error persist.

marcoojer, 2019-03-14:
Solved, I had a problem with my catkin_make that it was not compiling the iiwa_msgs package. Sorry my fault!! thanks anyway, I'll close the issue

closed 2019-03-14 by marcoojer