IFL-CAMP/iiwa_stack #158: My computer publish jointposition topic at a rate of 30hz, but sunrise can only receive 5hz. How can I improve control frequency?

lihx108, 2019-03-14:


lihx108, 2019-03-14:
[image]
[image]
I add some code in ROSSmartServo.java, and the rusult is the java side can not get all jointposition topic

lihx108, 2019-03-14:
@ @SalvoVirga Can you help me?

SalvoVirga, 2019-03-14:
Since `subscriber.getJointPosition()` returns null when there is no new message, the result so far is quite inconclusive.
Given that you send a message at a certain rate, you should check:
- the time delay between the reception of each message, which should optimally match the sending rate.
- if the order of the sent/received messages match

If you check that, please provide the full code that you use both on the Java side and the cpp/python one.
Optimally, this should be checked first on a toy example with a Sunrise Application that does just that for only one message subscriber, so that we can exclude any other source of error that might be there in ROSSmartServo.

Also, you should probably use the development branch and check all this there, the master branch is so behind and development will be (soon I hope :cry:) the new master. Master is so behind now that is doesn't make sense to test/fix this there.

lihx108, 2019-03-16:
> Since `subscriber.getJointPosition()` returns null when there is no new message, the result so far is quite inconclusive.
> Given that you send a message at a certain rate, you should check:
> 
> * the time delay between the reception of each message, which should optimally match the sending rate.
> * if the order of the sent/received messages match
> 
> If you check that, please provide the full code that you use both on the Java side and the cpp/python one.
> Optimally, this should be checked first on a toy example with a Sunrise Application that does just that for only one message subscriber, so that we can exclude any other source of error that might be there in ROSSmartServo.
> 
> Also, you should probably use the `development` branch and check all this there, the `master` branch is so behind and `development` will be (soon I hope ) the new `master`. `Master` is so behind now that is doesn't make sense to test/fix this there.

I have tryed development branch but there are many errors confused me. My PC send jointposition topic at a rate of 30hz, this is my ros side code: and the java side result is picture above, which is just 5hz. I have "rostopic echo iiwa/command/JointPosition" and "rostopic hz iiwa/command/JointPosition".and I find my pc published topic is OK. 

#include "beginner_tutorials/urmath.h"
#include "beginner_tutorials/inv_iiwa.h"
#include <math.h>
#include <stdio.h>
#include <fstream>
#include <iostream>
#include <vector>
#include <string>
#include "ros/ros.h"
#include "std_msgs/String.h"
#include "std_msgs/Int32.h"
#include "iiwa_msgs/JointPosition.h"
#include "iiwa_msgs/JointVelocity.h"
#include "iiwa_msgs/JointPositionVelocity.h"
#include "iiwa_msgs/ConfigureSmartServo.h"
#include "iiwa_msgs/SetPathParameters.h"
#include <sstream>
#include <trajectory_msgs/JointTrajectory.h>
#include <sensor_msgs/JointState.h>
#include "geometry_msgs/WrenchStamped.h"
#include "geometry_msgs/TwistStamped.h"
#include "geometry_msgs/PoseStamped.h"
#include "geometry_msgs/Twist.h"
#include "ros_igtl_bridge/igtltransform.h"
#include <time.h>
#include <chrono>
#include <Eigen/Eigen>

using namespace std;
using namespace Eigen;

double q_current[7]={0.0,0.0,0.0,0.0,0.0,0.0,0.0};
int posi_pub_count=0;
double joint_vel_last[] = {0.0,0.0,0.0,0.0,0.0,0.0,0.0};
double iiwa_quaternion[4] = {0.0,0.0,0.0,1.0};
double iiwa_position[3] = {0.0,0.0,0.0};
void iiwa_joint_position_Callback(const iiwa_msgs::JointPosition::ConstPtr& iiwa_jointmsg)
{
    q_current[0] = iiwa_jointmsg->position.a1;
    q_current[1] = iiwa_jointmsg->position.a2;
    q_current[2] = iiwa_jointmsg->position.a3;
    q_current[3] = iiwa_jointmsg->position.a4;
    q_current[4] = iiwa_jointmsg->position.a5;
    q_current[5] = iiwa_jointmsg->position.a6;
    q_current[6] = iiwa_jointmsg->position.a7;
}

int main(int argc, char **argv)
{
  ros::init(argc, argv, "iiwa_test");

  ros::NodeHandle n;

  ros::Publisher iiwa_joint_pub = n.advertise<iiwa_msgs::JointPosition>("/iiwa/command/JointPosition",1000);

  ros::Subscriber joint_po_sub = n.subscribe("/iiwa/state/JointPosition",1000,iiwa_joint_position_Callback);

[... 61 lines cut ...]
        iiwa_msgs::JointPosition JointPositionPubMsgs;
        JointPositionPubMsgs.header.stamp = ros::Time::now();
        JointPositionPubMsgs.header.seq = count;
        JointPositionPubMsgs.position.a1 = angle2_last(0)+0.17*sin(0.018*posi_pub_count);//angle2(0)
        JointPositionPubMsgs.position.a2 = angle2_last(1)+0.17*sin(0.018*posi_pub_count);//angle2(1)
        JointPositionPubMsgs.position.a3 = angle2_last(2)+0.17*sin(0.018*posi_pub_count);//angle2(2)
        JointPositionPubMsgs.position.a4 = angle2_last(3)+0.17*sin(0.018*posi_pub_count);//angle2(3)
        JointPositionPubMsgs.position.a5 = angle2_last(4)+0.17*sin(0.018*posi_pub_count);//angle2(4)
        JointPositionPubMsgs.position.a6 = angle2_last(5)+0.17*sin(0.018*posi_pub_count);//angle2(5)
        JointPositionPubMsgs.position.a7 = angle2_last(6)+0.17*sin(0.018*posi_pub_count);//angle2(6)

        iiwa_joint_pub.publish(JointPositionPubMsgs);
        posi_pub_count = posi_pub_count + 1;

    }
    ros::spinOnce();
    loop_rate.sleep();
    if(count >= 1900002000)
        count =2000;
    cout<<"count"<<count<<endl;
    count++;
  }
  return 0;
}
// %EndTag(FULLTEXT)%

lihx108, 2019-03-16:
@SalvoVirga Can you send me some test codes which make iiwa robot continous move and  publish topic on "iiwa/command/JointPosition"??

lihx108, 2019-03-18:
@SalvoVirgaI reset the IP,       C:\KUKA\Hardware\Manager\KUKAHardwareManager.exe -assign OptionNIC -os RTOS, and development branch can run now. I have test development branch, but it could not work more smoothly.  The iiwa only subscribe jointposition topic at  a rate of 5HZ.

lihx108, 2019-03-18:
@SalvoVirga  I reset the IP, C:\KUKA\Hardware\Manager\KUKAHardwareManager.exe -assign OptionNIC -os RTOS, and development branch can run now. I have test development branch, but it could not work more smoothly. The iiwa only subscribe jointposition topic at a rate of 5HZ.

lihx108, 2019-03-18:
#include <math.h>
#include <stdio.h>
#include <fstream>
#include <iostream>
#include <vector>
#include <string>
#include "ros/ros.h"
#include "std_msgs/String.h"
#include "std_msgs/Int32.h"
#include "iiwa_msgs/JointPosition.h"
#include "iiwa_msgs/JointVelocity.h"
#include "iiwa_msgs/JointPositionVelocity.h"
#include "iiwa_msgs/ConfigureSmartServo.h"
#include "iiwa_msgs/SetPathParameters.h"
#include <sstream>
#include <trajectory_msgs/JointTrajectory.h>
#include <sensor_msgs/JointState.h>
#include "geometry_msgs/WrenchStamped.h"
#include "geometry_msgs/TwistStamped.h"
#include "geometry_msgs/PoseStamped.h"
#include "geometry_msgs/Twist.h"
#include "ros_igtl_bridge/igtltransform.h"
#include <time.h>
#include <chrono>
#include <Eigen/Eigen>

using namespace std;
using namespace Eigen;

double q_current[7]={0.0,0.0,0.0,0.0,0.0,0.0,0.0};
int posi_pub_count=0;

double joint_vel_last[] = {0.0,0.0,0.0,0.0,0.0,0.0,0.0};
double iiwa_quaternion[4] = {0.0,0.0,0.0,1.0};
double iiwa_position[3] = {0.0,0.0,0.0};

void iiwa_joint_position_Callback(const iiwa_msgs::JointPosition::ConstPtr& iiwa_jointmsg)
{
    q_current[0] = iiwa_jointmsg->position.a1;
    q_current[1] = iiwa_jointmsg->position.a2;
    q_current[2] = iiwa_jointmsg->position.a3;
    q_current[3] = iiwa_jointmsg->position.a4;
    q_current[4] = iiwa_jointmsg->position.a5;
    q_current[5] = iiwa_jointmsg->position.a6;
    q_current[6] = iiwa_jointmsg->position.a7;

}

int main(int argc, char **argv)
{
  ros::init(argc, argv, "iiwa_test");
  ros::NodeHandle n;

  ros::Publisher iiwa_joint_pub = n.advertise<iiwa_msgs::JointPosition>("/iiwa/command/JointPosition",1000);

  ros::Subscriber joint_po_sub = n.subscribe("/iiwa/state/JointPosition",1000,iiwa_joint_position_Callback);

  ros::ServiceClient control_mode_client = n.serviceClient<iiwa_msgs::ConfigureSmartServo>("/iiwa/configuration/ConfigureSmartServo");

  ros::Rate loop_rate(30);//125

  ros::Duration(1.5).sleep();
  int count=0;

  Eigen::VectorXd angle2_last(7);
  angle2_last<<0,0,0,0,0,0,0;

  iiwa_msgs::ConfigureSmartServo iiwa_config_srv_5;
  iiwa_config_srv_5.request.control_mode = 1;
  iiwa_config_srv_5.request.joint_impedance.joint_stiffness.a1 = 200;
  iiwa_config_srv_5.request.joint_impedance.joint_stiffness.a2 = 200;
  iiwa_config_srv_5.request.joint_impedance.joint_stiffness.a3 = 200;
  iiwa_config_srv_5.request.joint_impedance.joint_stiffness.a4 = 200;
  iiwa_config_srv_5.request.joint_impedance.joint_stiffness.a5 = 200;
  iiwa_config_srv_5.request.joint_impedance.joint_stiffness.a6 = 200;
  iiwa_config_srv_5.request.joint_impedance.joint_stiffness.a7 = 200;

  iiwa_config_srv_5.request.joint_impedance.joint_damping.a1 = 0.3;
  iiwa_config_srv_5.request.joint_impedance.joint_damping.a2 = 0.3;
  iiwa_config_srv_5.request.joint_impedance.joint_damping.a3 = 0.3;
  iiwa_config_srv_5.request.joint_impedance.joint_damping.a4 = 0.3;
  iiwa_config_srv_5.request.joint_impedance.joint_damping.a5 = 0.3;
  iiwa_config_srv_5.request.joint_impedance.joint_damping.a6 = 0.3;
  iiwa_config_srv_5.request.joint_impedance.joint_damping.a7 = 0.3;
  int  posi_pub_count = 0;

  while (ros::ok())
  {
        if(count == 90)
        {
            iiwa_config_srv_5.request.joint_impedance.joint_stiffness.a1 = 1000;
            iiwa_config_srv_5.request.joint_impedance.joint_stiffness.a2 = 1000;
            iiwa_config_srv_5.request.joint_impedance.joint_stiffness.a3 = 1000;
            iiwa_config_srv_5.request.joint_impedance.joint_stiffness.a4 = 1000;
            iiwa_config_srv_5.request.joint_impedance.joint_stiffness.a5 = 1000;
            iiwa_config_srv_5.request.joint_impedance.joint_stiffness.a6 = 1000;
            iiwa_config_srv_5.request.joint_impedance.joint_stiffness.a7 = 1000;

            if (control_mode_client.call(iiwa_config_srv_5)) {
                        if(!iiwa_config_srv_5.response.success)
                            ROS_ERROR_STREAM("Config failed, Java error: " << iiwa_config_srv_5.response.error);
                        else
                            ROS_INFO_STREAM("SmartServo Service successfully called.");
                 }
            else {
                       ROS_ERROR_STREAM("Config failed - service could not be called - QUITTING NOW !");
                 }

         angle2_last(0) = q_current[0];
         angle2_last(1) = q_current[1];
         angle2_last(2) = q_current[2];
         angle2_last(3) = q_current[3];
         angle2_last(4) = q_current[4];
         angle2_last(5) = q_current[5];
         angle2_last(6) = q_current[6];

        }

    if (count>100)
    {
        iiwa_msgs::JointPosition JointPositionPubMsgs;
        JointPositionPubMsgs.header.stamp = ros::Time::now();
        JointPositionPubMsgs.header.seq = count;
        JointPositionPubMsgs.position.a1 = angle2_last(0)+0.17*sin(0.018*posi_pub_count);//angle2(0)
        JointPositionPubMsgs.position.a2 = angle2_last(1)+0.17*sin(0.018*posi_pub_count);//angle2(1)
        JointPositionPubMsgs.position.a3 = angle2_last(2)+0.17*sin(0.018*posi_pub_count);//angle2(2)
        JointPositionPubMsgs.position.a4 = angle2_last(3)+0.17*sin(0.018*posi_pub_count);//angle2(3)
        JointPositionPubMsgs.position.a5 = angle2_last(4)+0.17*sin(0.018*posi_pub_count);//angle2(4)
        JointPositionPubMsgs.position.a6 = angle2_last(5)+0.17*sin(0.018*posi_pub_count);//angle2(5)
        JointPositionPubMsgs.position.a7 = angle2_last(6)+0.17*sin(0.018*posi_pub_count);//angle2(6)

        iiwa_joint_pub.publish(JointPositionPubMsgs);
        posi_pub_count = posi_pub_count + 1;

    }
    ros::spinOnce();
    loop_rate.sleep();
    if(count >= 1900002000)
        count =2000;
    cout<<"count"<<count<<endl;
    count++;

  }
  return 0;
}

lihx108, 2019-03-18:
KUKA SUNRISE version: 1.15

JimmyDaSilva, 2020-02-25:
@lihx108 Did you find a way around this ?
I am having exactly the same problem.