IFL-CAMP/iiwa_stack #160: Inverse Kinematics

lin2jy, 2019-03-25:
Which method of inverse kinematics and what optimization criteria are used in iiwa_stack?

SalvoVirga, 2019-03-26:
iiwa_stack provides a ROS interface to the KUKA APIs, with support for RViz, MoveIt!, etc.
So it doesn't use IK or optimization methods per se. If you refer to the fact that we provide an interface to command the robot is Cartesian space, that uses the KUKA APIs as well, so whatever is used there is proprietary KUKA stuff.
Using MoveIt! instead, you can use any method that is supported there (IKFast, TRAC-IK, ...)

closed 2019-04-04 by SalvoVirga

Deastan, 2019-06-18:
Hello, 

To stay one the same topic, I ask here a question. Which solver did you use ? Do you have a preferences ? Did you try IKFast or TRAC-IK? 
I am using the default one RRTkConfigDefault... But i think IKFast is better and faster.
Best,

John