IFL-CAMP/iiwa_stack #161: No p gain specified for pid

westinsykes, 2019-04-03:
I'm getting this issue where it cannot find p gains it wants and I'm wondering if it is because I'm using newer gazebo. I have a clean installation of Melodic, although I was seeing this on Kinetic as well. I'm unsure whether the depricated syntax warnings are related, although they always seem to come together.

> <pre><font color="#729FCF"><b>~/catkin_ws</b></font>$ roslaunch iiwa_gazebo iiwa_world.launch 
> ... logging to /home/rosseymour/.ros/log/4c4493d2-564a-11e9-9aa5-e0d55e2b0377/roslaunch-rosseymour-X399-DESIGNARE-EX-53369.log
> Checking log directory for disk usage. This may take awhile.
> Press Ctrl-C to interrupt
> Done checking log file disk usage. Usage is &lt;1GB.
> 
> <b>started roslaunch server http://rosseymour-X399-DESIGNARE-EX:42833/</b>
> 
> SUMMARY
> ========
> 
> PARAMETERS
>  * /gazebo/enable_ros_network: True
>  * /robot_description: &lt;?xml version=&quot;1....
>  * /rosdistro: melodic
>  * /rosversion: 1.14.3
>  * /use_sim_time: True
> 
> NODES
>   /
>     gazebo (gazebo_ros/gzserver)
>     gazebo_gui (gazebo_ros/gzclient)
>     urdf_spawner (gazebo_ros/spawn_model)
> 
> <b>ROS_MASTER_URI=http://localhost:11311</b>
> 
> <b>process[gazebo-1]: started with pid [53387]</b>
> <b>process[gazebo_gui-2]: started with pid [53392]</b>
> <b>process[urdf_spawner-3]: started with pid [53397]</b>
> [ INFO] [1554327009.624922581]: Finished loading Gazebo ROS API Plugin.
> [ INFO] [1554327009.625826695]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
> [ INFO] [1554327009.638017407]: Finished loading Gazebo ROS API Plugin.
> [ INFO] [1554327009.639311464]: waitForService: Service [/gazebo_gui/set_physics_properties] has not been advertised, waiting...
> [INFO] [1554327010.027814, 0.000000]: Loading model XML from ros parameter robot_description
> [INFO] [1554327010.038934, 0.000000]: Waiting for service /gazebo/spawn_urdf_model
> [ INFO] [1554327010.383950067, 0.022000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
> [ INFO] [1554327010.403749606, 0.041000000]: Physics dynamic reconfigure ready.
> [INFO] [1554327010.643423, 0.276000]: Calling service /gazebo/spawn_urdf_model
> [INFO] [1554327010.787069, 0.356000]: Spawn status: SpawnModel: Successfully spawned entity
> <font color="#EF2929"><b>[Err] [REST.cc:205] Error in REST request</b></font>
> 
> libcurl: (51) SSL: no alternative certificate subject name matches target host name &apos;api.ignitionfuel.org&apos;
> [ INFO] [1554327010.878234699, 0.356000000]: Loading gazebo_ros_control plugin
> [ INFO] [1554327010.878314040, 0.356000000]: Starting gazebo_ros_control plugin in namespace: /iiwa
> [ INFO] [1554327010.878878606, 0.356000000]: gazebo_ros_control plugin is waiting for model URDF in parameter [/robot_description] on the ROS param server.
> <font color="#C4A000">[ WARN] [1554327010.991328540, 0.356000000]: Deprecated syntax, please prepend &apos;hardware_interface/&apos; to &apos;PositionJointInterface&apos; within the &lt;hardwareInterface&gt; tag in joint &apos;iiwa_joint_1&apos;.</font>
> <font color="#CC0000">[ERROR] [1554327010.992209951, 0.356000000]: No p gain specified for pid.  Namespace: /iiwa/gazebo_ros_control/pid_gains/iiwa_joint_1</font>
> <font color="#C4A000">[ WARN] [1554327010.992250618, 0.356000000]: Deprecated syntax, please prepend &apos;hardware_interface/&apos; to &apos;PositionJointInterface&apos; within the &lt;hardwareInterface&gt; tag in joint &apos;iiwa_joint_2&apos;.</font>
> <font color="#CC0000">[ERROR] [1554327010.992799324, 0.356000000]: No p gain specified for pid.  Namespace: /iiwa/gazebo_ros_control/pid_gains/iiwa_joint_2</font>
> <font color="#C4A000">[ WARN] [1554327010.992838659, 0.356000000]: Deprecated syntax, please prepend &apos;hardware_interface/&apos; to &apos;PositionJointInterface&apos; within the &lt;hardwareInterface&gt; tag in joint &apos;iiwa_joint_3&apos;.</font>
> <font color="#CC0000">[ERROR] [1554327010.993350585, 0.356000000]: No p gain specified for pid.  Namespace: /iiwa/gazebo_ros_control/pid_gains/iiwa_joint_3</font>
> <font color="#C4A000">[ WARN] [1554327010.993383057, 0.356000000]: Deprecated syntax, please prepend &apos;hardware_interface/&apos; to &apos;PositionJointInterface&apos; within the &lt;hardwareInterface&gt; tag in joint &apos;iiwa_joint_4&apos;.</font>
> <font color="#CC0000">[ERROR] [1554327010.993885616, 0.356000000]: No p gain specified for pid.  Namespace: /iiwa/gazebo_ros_control/pid_gains/iiwa_joint_4</font>
> <font color="#C4A000">[ WARN] [1554327010.993914901, 0.356000000]: Deprecated syntax, please prepend &apos;hardware_interface/&apos; to &apos;PositionJointInterface&apos; within the &lt;hardwareInterface&gt; tag in joint &apos;iiwa_joint_5&apos;.</font>
> <font color="#CC0000">[ERROR] [1554327010.994442357, 0.356000000]: No p gain specified for pid.  Namespace: /iiwa/gazebo_ros_control/pid_gains/iiwa_joint_5</font>
> <font color="#C4A000">[ WARN] [1554327010.994470300, 0.356000000]: Deprecated syntax, please prepend &apos;hardware_interface/&apos; to &apos;PositionJointInterface&apos; within the &lt;hardwareInterface&gt; tag in joint &apos;iiwa_joint_6&apos;.</font>
> <font color="#CC0000">[ERROR] [1554327010.994998788, 0.356000000]: No p gain specified for pid.  Namespace: /iiwa/gazebo_ros_control/pid_gains/iiwa_joint_6</font>
> <font color="#C4A000">[ WARN] [1554327010.995027563, 0.356000000]: Deprecated syntax, please prepend &apos;hardware_interface/&apos; to &apos;PositionJointInterface&apos; within the &lt;hardwareInterface&gt; tag in joint &apos;iiwa_joint_7&apos;.</font>
> <font color="#CC0000">[ERROR] [1554327010.995552053, 0.356000000]: No p gain specified for pid.  Namespace: /iiwa/gazebo_ros_control/pid_gains/iiwa_joint_7</font>
> [ INFO] [1554327010.999307413, 0.356000000]: Loaded gazebo_ros_control.
> <b>[urdf_spawner-3] process has finished cleanly</b>
> </pre>

Deastan, 2019-04-09:
Same issue... Do someone have a solution ? Did you find a solution @westinsykes ?

`jonathan@jonathan:~/catkin_ws$ roslaunch iiwa_gazebo  iiwa_gazebo.launch model:=iiwa7
... logging to /home/jonathan/.ros/log/d0164b2e-5b09-11e9-9243-e4a47163547f/roslaunch-jonathan-csem-14434.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: Traditional processing is deprecated. Switch to --inorder processing!
To check for compatibility of your document, use option --check-order.
For more infos, see http://wiki.ros.org/xacro#Processing_Order
started roslaunch server http://jonathan-csem:38297/

SUMMARY
========

PARAMETERS
 * /iiwa/EffortJointInterface_J1_controller/joint: iiwa_joint_1
 * /iiwa/EffortJointInterface_J1_controller/pid/d: 80.0
 * /iiwa/EffortJointInterface_J1_controller/pid/i: 100
 * /iiwa/EffortJointInterface_J1_controller/pid/i_clamp_max: 10000
 * /iiwa/EffortJointInterface_J1_controller/pid/i_clamp_min: -10000
 * /iiwa/EffortJointInterface_J1_controller/pid/p: 800.0
 * /iiwa/EffortJointInterface_J1_controller/type: effort_controller...
 * /iiwa/EffortJointInterface_J2_controller/joint: iiwa_joint_2
 * /iiwa/EffortJointInterface_J2_controller/pid/d: 100.0
 * /iiwa/EffortJointInterface_J2_controller/pid/i: 1000
 * /iiwa/EffortJointInterface_J2_controller/pid/i_clamp_max: 10000
 * /iiwa/EffortJointInterface_J2_controller/pid/i_clamp_min: -10000
 * /iiwa/EffortJointInterface_J2_controller/pid/p: 800.0
 * /iiwa/EffortJointInterface_J2_controller/type: effort_controller...
 * /iiwa/EffortJointInterface_J3_controller/joint: iiwa_joint_3
 * /iiwa/EffortJointInterface_J3_controller/pid/d: 5.0
 * /iiwa/EffortJointInterface_J3_controller/pid/i: 10
 * /iiwa/EffortJointInterface_J3_controller/pid/i_clamp_max: 10000
 * /iiwa/EffortJointInterface_J3_controller/pid/i_clamp_min: -10000
 * /iiwa/EffortJointInterface_J3_controller/pid/p: 800.0
 * /iiwa/EffortJointInterface_J3_controller/type: effort_controller...
 * /iiwa/EffortJointInterface_J4_controller/joint: iiwa_joint_4
 * /iiwa/EffortJointInterface_J4_controller/pid/d: 80.0
 * /iiwa/EffortJointInterface_J4_controller/pid/i: 10
 * /iiwa/EffortJointInterface_J4_controller/pid/i_clamp_max: 10000
 * /iiwa/EffortJointInterface_J4_controller/pid/i_clamp_min: -10000
 * /iiwa/EffortJointInterface_J4_controller/pid/p: 800.0
 * /iiwa/EffortJointInterface_J4_controller/type: effort_controller...
 * /iiwa/EffortJointInterface_J5_controller/joint: iiwa_joint_5
 * /iiwa/EffortJointInterface_J5_controller/pid/d: 10.0
 * /iiwa/EffortJointInterface_J5_controller/pid/i: 5
 * /iiwa/EffortJointInterface_J5_controller/pid/i_clamp_max: 10000
 * /iiwa/EffortJointInterface_J5_controller/pid/i_clamp_min: -10000
 * /iiwa/EffortJointInterface_J5_controller/pid/p: 300.0
 * /iiwa/EffortJointInterface_J5_controller/type: effort_controller...
 * /iiwa/EffortJointInterface_J6_controller/joint: iiwa_joint_6
 * /iiwa/EffortJointInterface_J6_controller/pid/d: 1.0
 * /iiwa/EffortJointInterface_J6_controller/pid/i: 1
 * /iiwa/EffortJointInterface_J6_controller/pid/i_clamp_max: 10000
 * /iiwa/EffortJointInterface_J6_controller/pid/i_clamp_min: -10000
 * /iiwa/EffortJointInterface_J6_controller/pid/p: 50.0
 * /iiwa/EffortJointInterface_J6_controller/type: effort_controller...
 * /iiwa/EffortJointInterface_J7_controller/joint: iiwa_joint_7
 * /iiwa/EffortJointInterface_J7_controller/pid/d: 1.0
 * /iiwa/EffortJointInterface_J7_controller/pid/i: 0.1
 * /iiwa/EffortJointInterface_J7_controller/pid/i_clamp_max: 10000
 * /iiwa/EffortJointInterface_J7_controller/pid/i_clamp_min: -10000
 * /iiwa/EffortJointInterface_J7_controller/pid/p: 10.0
 * /iiwa/EffortJointInterface_J7_controller/type: effort_controller...
 * /iiwa/EffortJointInterface_trajectory_controller/action_monitor_rate: 30
 * /iiwa/EffortJointInterface_trajectory_controller/constraints/goal_time: 0.5
 * /iiwa/EffortJointInterface_trajectory_controller/gains/iiwa_joint_1/d: 30
 * /iiwa/EffortJointInterface_trajectory_controller/gains/iiwa_joint_1/i: 15
[... 158 lines cut ...]
[ WARN] [1554843335.310162864, 0.164000000]: Deprecated syntax, please prepend 'hardware_interface/' to 'PositionJointInterface' within the <hardwareInterface> tag in joint 'iiwa_joint_7'.
[ERROR] [1554843335.311026535, 0.164000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/iiwa_joint_7
[ INFO] [1554843335.315603488, 0.164000000]: Loaded gazebo_ros_control.
[ WARN] [1554843335.315838278, 0.165000000]: The default_robot_hw_sim plugin is using the Joint::SetPosition method without preserving the link velocity.
[ WARN] [1554843335.315863262, 0.165000000]: As a result, gravity will not be simulated correctly for your model.
[ WARN] [1554843335.315880030, 0.165000000]: Please set gazebo_pid parameters, switch to the VelocityJointInterface or EffortJointInterface, or upgrade to Gazebo 9.
[ WARN] [1554843335.315895683, 0.165000000]: For details, see https://github.com/ros-simulation/gazebo_ros_pkgs/issues/612
[ INFO] [1554843335.337601045, 0.186000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[urdf_spawner-4] process has finished cleanly
log file: /home/jonathan/.ros/log/d0164b2e-5b09-11e9-9243-e4a47163547f/urdf_spawner-4*.log
[ INFO] [1554843335.366359394, 0.215000000]: Physics dynamic reconfigure ready.
[INFO] [1554843335.499335, 0.347000]: Controller Spawner: Waiting for service controller_manager/switch_controller
[... cut ...] ositionJointInterface_J4_controller, PositionJointInterface_J5_controller, PositionJointInterface_J6_controller, PositionJointInterface_J7_controller
[INFO] [1554843335.557224, 0.406000]: Started controllers: joint_state_controller, PositionJointInterface_J1_controller, PositionJointInterface_J2_controller, PositionJointInterface_J3_controller, PositionJointInterface_J4_controller, PositionJointInterface_J5_controller, PositionJointInterface_J6_controller, PositionJointInterface_J7_controller

`

westinsykes, 2019-04-09:
I didn't find a solution I liked. I think the simulator doesn't work correctly on Kinetic. I installed Ubuntu 14.04 and ROS Indigo and now it works. Unfortunately Indigo is at the end of its life.

Deastan, 2019-04-11:
And do you know if for the communication with the kuka robot works on kinetic well ? I saw that it should! Best,

westinsykes, 2019-04-12:
Sorry, I haven't tried. I assume it does.

<mention@noreply.github.com>
Date:   04/11/2019 06:45 AM
(#161)

[EXTERNAL]

ymollard, 2019-04-24:
I fixed the IIWA Gazebo simulation in Melodic that way: https://github.com/ymollard/iiwa_stack/commit/e9312df7b9a87f2f68e2fb56610816e117bdde38

I've tested different gains, see the dev branch: https://github.com/ymollard/iiwa_stack/commit/0e3db65651775d395542ce0809c1210a067346f2

I'm not satisfied with these gains, though. There's still a bit of bouncing. If someone has a little bit of time to tweak PID gains that would be a nice fix. I assume that in previous versions it was using the Gazebo default

Deastan, 2019-06-05:
Thank you!
Do you think with the addition of the PID gains in the iiwa_control.yaml makes it works also for Kenetic?  On which Gazebo version did you try ? 
Is it working with melodic on the real robot ? Because if everything work, I might move on it.
Best,
Jonathan

ymollard, 2019-06-05:
I've runned this on melodic with Gazebo 9.0.0.
I haven't tested on other versions, you can make the test on kinetic.
AFAIK, these gains are only used by Gazebo, the real robot must have other PID control elsewhere. I've haven't tested this `iiwa_stack` package on a real robot because I miss the smart servo plugin for Sunrise Workbench. I'm using the FRI instead with [`iiwa_ros`](https://github.com/ymollard/iiwa_ros) which runs fine on the real robot with melodic.

westinsykes, 2019-06-05:
I am out of the office until 06/12/2019.

Note: This is an automated response to your message  "Re:
[IFL-CAMP/iiwa_stack] No p gain specified for pid (#161)" sent on 6/5/2019
4:39:51 AM.

This is the only notification you will receive while this person is away.

westinsykes, 2019-06-05:
I am out of the office until 06/12/2019.

Note: This is an automated response to your message  "Re:
[IFL-CAMP/iiwa_stack] No p gain specified for pid (#161)" sent on 6/5/2019
3:48:28 AM.

This is the only notification you will receive while this person is away.

Deastan, 2019-07-03:
Thank for your information ! I will test one day the one from EPFL but I think it is really close to this one.

ymollard@0e3db65 => I alsi so this commit and this one: commit e9312df7b9a87f2f68e2fb56610816e117bdde38

How can I be sure that it work ? Because I have added you modif but it seems that it doesn't work...
Because for me, the end-effector still move a lot.. (I am still on gazebo 2.2). Do you think we can improve it ?