IFL-CAMP/iiwa_stack #164: Rviz examples not working on Ubuntu 18.04 + ROS Melodic

CKowal, 2019-04-15:
OS: Ubuntu 18.04
ROS: Melodic
Short error description: iiwa representation (e.g. by using "roslaunch iiwa_moveit moveit_planning_execution.launch" in Rviz not working, various topics not subscribed, Gazebo state not read properly.
[image]

Tried several example files but Rviz always fails to setup correctly. Tried both the master and the development branch after a clean install. Is this a common known issue in Melodic?

Warnings & Errors during the launch:
- [ WARN] [1555322408.268605456, 0.342000000]: Deprecated syntax, please prepend 'hardware_interface/' to 'PositionJointInterface' within the <hardwareInterface> tag in joint 'iiwa_joint_1'.
[ERROR] [1555322408.270399535, 0.342000000]: No p gain specified for pid.  Namespace: /iiwa/gazebo_ros_control/pid_gains/iiwa_joint_1
...
- [ WARN] [1555322411.222563568, 2.712000000]: Kinematics solver doesn't support #attempts anymore, but only a timeout.
Please remove the parameter '/iiwa/rviz_nboffis104_18543_3933730925375637416/manipulator/kinematics_solver_attempts' from your configuration.

Thanks in advance for any useful information.

matthiashh, 2019-04-15:
The first warning is tackled in #84. The error in #161 

About the second warning I am not sure if older versions of ROS that are still widely used would be compatible if it would be removed.

But apart from that I had issues with collapsing RViz representations as well and I solved them by putting
```
# RViz Workaround
# Otherwise the robot model will collapse
export LC_NUMERIC="en_US.UTF-8"
```
in my `.bashrc`. I am on melodic and it works for me.

CKowal, 2019-04-15:
Wow, this workaround indeed did the job. Can you also explain why this .bashrc addon has an impact on the collapsed robo in Rviz? I'm just curious as I can't draw any connection.

matthiashh, 2019-04-16:
It doesn't fully apply for me, but this issue seems to be related:
https://github.com/ros-planning/moveit/issues/1050

CKowal, 2019-04-16:
Thanks alot!

closed 2019-04-16 by CKowal