IFL-CAMP/iiwa_stack #175: Robot model is messed up in RVIZ / melodic

exo-core, 2019-05-24:
Hi all,

I just tried to run our existing robot URDF in a melodic workspace. When launching our MoveIt! configuration (`roslaunch tum_iiwa_zimmer_r840_moveit moveit_planning_execution.launch`) the robot looks fine in Gazebo:

[image]

Also the TF tree is looking fine to me. However all 3d markers and meshes in our worcell are just randomly lying on the floor:

[image]

Did anybody experience a similar issue and has a clue where it is coming from?

Ideas & Observations:
*  We already tried to set-up a new MoveIt! description but the problem was the same
*  Could it be an issue that the `joint_state` of our gripper fingers is not available in the simulation (in kinetic RVIZ just did not show the fingers...)?
*  Do I need to update something due to TF2?

matthiashh, 2019-05-24:
Does the solution of #164  work for you?

exo-core, 2019-05-24:
Indeed it does! Thanks! :smiley:

closed 2019-05-24 by exo-core