IFL-CAMP/iiwa_stack #179: Changing desired_force in DesiredForceControlMode has no impact

CKowal, 2019-06-06:
Hey everyone,

changing the desired force from e.g. 80 N to 150 N in DesiredForceControlMode has no impact - the robot is not pushing with more force. I couldn't find any code that specifies a maximum force. Even when the safety configuration limits are disabled, the robot pushes with the same force after increasing the desired force.
Does anyone have an idea where this could come from? Thanks in advance.

SalvoVirga, 2019-06-06:
Well, did you try with lower forces? e.g. 1N to 5N?

That should work, I never tried to use forces higher than 20N. Logically, I imagine there might be an upper limit, but I don't recall it from the documentation.
Please try with lower forces first, the title "Changing desired_force has no impact" could be misleading if only very high forces are not available.

CKowal, 2019-06-06:
Thanks for the response, I will check it tomorrow morning and give an update.

Could you image where this upper limit might come from? Could it be configurable?

SalvoVirga, 2019-06-07:
From the KUKA APIs documentation

`The robot will perform the pattern or create the desired force only as long as the robot is inside it's specified position limit (see positionLimit(double, double, double, double, double, double)). `

and 

`The default values of the position limit are infinity. `

Also checked and this is the case also for the settings used in iiwa_stack, so there should not be an issue from our side. Maybe there is an undocumented limit set by KUKA, not sure.

SalvoVirga, 2019-06-14:
any update on this?

CKowal, 2019-06-17:
I have checked the safety configuration and couldn't find any limit in there. Still have to measure the applied force with some kind of scale in order to be really sure that there is no difference between 75 N and 150 N for example. Will update as soon as I've done that.

JimmyDaSilva, 2020-02-11:
As the max payload for the iiwa 7 and 14 are respectively 7kg and 14kg in the most favorable configuration, I guess the max forces it can apply with no tool attached is around 70N, and 140N.

To confirm that, one could compute the max force in a particular position with the max alowed torques.