IFL-CAMP/iiwa_stack #180: Error: Group 'manipulator' was not found.

Xend1992, 2019-06-10:
Hi,

I am currently working on writing a 'pick and place' script using your iiwa14 urdf, but I happen to see some problem which will not allow me to move anywhere. I will add that I recently started using ROS, so not everything is clear to me.

I am running this command to bring up RVIZ and Gazebo simulations:
`roslaunch iiwa_moveit moveit_planning_execution.launch`

Everything loads without any problems. I can plan/execute different paths in Rviz, and then see them in Gazebo. When I try to do it programmatically (from node written in python), I get this error:

`Group 'manipulator' was not found`

I can see that the planning group is named **manipulator** in Rviz, but somehow my node can't see this planning group (or there is a problem of other nature).

The piece of code that is causing the error is below:
```
if __name__ == '__main__':
    moveit_commander.roscpp_initialize(sys.argv)
    rospy.init_node('pick_place_demo', anonymous=True)

    robot = moveit_commander.RobotCommander()

    scene = moveit_commander.PlanningSceneInterface()

    group_name = "manipulator"
    group = moveit_commander.MoveGroupCommander(group_name)
    

    rospy.spin()
```

I am running this on Ubuntu 16.04 with ROS Kinetic. 

Am I missing something here? How could I solve this problem?
Any help is appreciated,
Thanks

SalvoVirga, 2019-06-10:
Check if that planning group lives under the "iiwa" namespace, if so you could just run also your node under that ROS namespace.

Xend1992, 2019-06-11:
Thanks for your quick answer.

You're right! The planning group lives under 'iiwa' planning group and creating a launch file that runs my node in 'iiwa' namespace fixed the error. I believe I can close the issue now.

Thanks for your help

closed 2019-06-11 by Xend1992

ar-i-enterprise, 2020-05-22:
> Thanks for your quick answer.
> 
> You're right! The planning group lives under 'iiwa' planning group and creating a launch file that runs my node in 'iiwa' namespace fixed the error. I believe I can close the issue now.
> 
> Thanks for your help

hai can you please explain me in details on how did you actually resolve this error?  I cant understand with the term of "lives under iiwa"

n3qu, 2020-09-17:
> Thanks for your quick answer.
> 
> You're right! The planning group lives under 'iiwa' planning group and creating a launch file that runs my node in 'iiwa' namespace fixed the error. I believe I can close the issue now.
> 
> Thanks for your help

Hi, could you go into more specifics about what you did to solve this issue? Thanks!

mf093087, 2022-01-19:
something like this ?          arm = MoveGroupCommander('iiwa\manipulator')