IFL-CAMP/iiwa_stack #181: Question: buttonEvent

emielke12, 2019-06-11:
what does the `buttonEvent` topic do? Is this related to the flange button on some iiwas? If so, does pressing the button trigger anything? 

I would like to set up the button to enable certain control modes through the rosservice pipeline, and I am wondering if the buttonEvent would be a way to allow me to do that? Or would I have to do something on the sunrise side of things?

SalvoVirga, 2019-06-13:
Hum, not!
The `buttonEvent` topic is part of an interface that lets automatically configure a **UserKey** menu on the **SmartPad** from a ROS parameter. A **UserKey** menu uses those 4 blank buttons on the bottom left of the **SmartPad**, the status of those buttons is published as events on that ROS topic.
This functionality is a bit hidden and (sadly) not well documented, you can check `setupToolbars` and `getToolbarSpecifications` in `Configuration.java`.

But!
With #108 it was introduced a way to publish the state of the green button on the **MediaFlange**. You can check the changes provided by that PR. Those changes are integrated in iiwa_stack but they are commented by default, since not all iiwa's come with that flange and the project would not compile if the selected iiwa is not configured with a **MediaFlange**.
You can search for `MEDIAFLANGEIO` in `iiwaPublisher.java` and `ROSBaseApplication.java` and uncomment the marked lines to enable the ROS topic publishing the button state. Then from callbacks to that topics you can do whatever you want :)

emielke12, 2019-06-25:
Great, thanks for the info!

emielke12, 2019-07-22:
I'm just getting around to trying this out, and I came across a few errors when uncommenting the `MEDIAFLANGEIO` portions. 

 1. I get an error on this line `flangeButton.setData(mediaFlange.getUserButton());` in the `iiwaPublisher.java` file, indicating there is no `getUserButton` method for the object. In looking at available methods, it looks like there are `getInput#` methods for #s 1-8.

 2. Once the `iiwaPublisher.java` media flange portions have been uncommented, an error appears in `PublisherThread.java` on the line `publisher.publishCurrentState(endpointFrame)`. This error indicates it it is looking for a MediaFlangeIOGroup object to be passed in as an argument. 

Just wondering if you have seen these bugs.

exo-core, 2019-07-23:
I ran into the same issue. It seems that that same line got missing when merging everything together in the 'new' `master` branch. I resolved the issue by passing a reference to the `MediaFlangeIO` to the `iiwaPublisher`'s constructor.
If I find the time I'll create a PR for the fix when I'm in the office later on...

exo-core, 2019-07-26:
Fixed it in #199.

emielke12, 2019-08-07:
@exo-core I applied #199 and uncommented the MediaFlangeIO portions. The errors disappeard, but the corresponding topic for the flange didn't appear in the published topics. Are you able to see the topic using the changes?

emielke12, 2019-08-08:
UPDATE: I realized I didn't comment out the portion to `@Inject` in `RosBaseApplication.java`. I am now seeing the Media Flange topic.

However, when I try to apply the status of the button to `flangeButton` in line `// flangeButton.setData(((MediaFlangeIOGroup)mediaFlange).getUserButton()); // MEDIAFLANGEIO` I get an error. Looking at the class, there is no method `getUserButton`.

I attempted to work around this by using `getInput("UserButton")` and casting to boolean, but that just displayed an `IllegalArgumentException` when running the application and trying to echo the topic. 

Any ideas here?

emielke12, 2019-08-09:
Fixed it. For some reason the work visual file in my project was not recognizing the media flange I was using. Remade the project and it had the correct flange and I could use `getUserButton()` as a method in the `MediaFlangeIOGroup` class. Sorry for the confusion.

closed 2019-08-09 by emielke12

exo-core, 2019-08-13:
Hi @emielke12,

the method exists and works fine at our end:

[image]

Are you sure you selected the correct media flange (`Touch Electric` or `Touch Pneumatic`)  when creating the Sunrise project?

[image]