IFL-CAMP/iiwa_stack #182: Sunrise project setup errors

isabellahuang, 2019-06-12:
I'm following the steps [here ](https://github.com/IFL-CAMP/iiwa_stack/wiki/sunrise_project_setup)to set up the Sunrise project to use iiwa_stack. However, I have an error (Sunrise Workbench v 1.10) and I was just wondering if removing the hint parameter to fix the problem is fine. Thanks!

[image]

SalvoVirga, 2019-06-13:
The error shows that the version of rosjava that you have in your build path is not the correct one. Be sure to remove the current libraries in the **ROSJavaLib** folder, delete the folder and then copy/link the one that comes with the latest version of the code.

isabellahuang, 2019-06-13:
Thank you, I made a fresh project and refreshed the libraries. It works now!

closed 2019-06-13 by isabellahuang