IFL-CAMP/iiwa_stack #183: ROSSmartServo error, java.net.BindException: Address already in use

isabellahuang, 2019-06-12:
I'm trying to run the ROSSmartServo program but it is unable to connect to the ROS master. I've attached a picture of the error message I receive. Any tips why this might be the case? The IP address is set correctly and I can ping the controller just fine.

exo-core, 2019-06-13:
Just restart the controller by powering it off and on again.

matthiashh, 2019-06-13:
> Just restart the controller by powering it off and on again.

That's also what I did up to now when having this error message. It felt like a quite unsatisfactory solution though.

SalvoVirga, 2019-06-13:
Yes, that is a known issue and restarting the cabinet is the only solution at the moment :\

From our experience, the error appears only after _some event_ leads to a not correct setup of the communication with the ROS Master and consequent exceptions are thrown.
In practice if `ROSSmartServo` starts and stops correctly, this error will not appear.

This should show that the `dispose()` function in `ROSBaseApplication.java` does its job correctly, everything it disposed when the application quits. This function is an overload from the base class `RoboticsAPIApplication` that should be automatically called internally by KUKA. My understanding is that in some cases this doesn't happen, or something else prevents resources (in our case a socket) to be released correctly.

I will try to at least find a way to replicate the error in structured manner so that we can test solutions, but it will be painful since one needs to restart the cabinet every time the solution doesn't work :D

isabellahuang, 2019-06-13:
Thank you so much everyone, this is super helpful. A restart didn't work for me this time, but once I fixed my issue posted in #182, it worked.

matthiashh, 2019-06-13:
Should we keep this open since this error still arises in other configurations?
I had quite a lot of these before I figured out a stable setup.

BrunoScaglioni, 2019-06-14:
> From our experience, the error appears only after _some event_ leads to a not correct setup of the communication with the ROS Master and consequent exceptions are thrown.
> In practice if `ROSSmartServo` starts and stops correctly, this error will not appear.

Probably a trivial question but I can't figure it out: How is it possible to stop the application correctly? From the smartPad it is possible to stop it and reset it, but reset throws an exception (because it is interrupted), and the app is an infinite loop, as it should be.  
Is there another way?

SalvoVirga, 2019-06-17:
We usually have the following to start everything:
- run a `roscore` on a terminal that stays up at all times
- optionally, run other `rosrun` or `roslaunch` from other terminals as needed
- Start `ROSSMartServo` on the `SmartPad`

And to stop:
- Press the `Stop Button` (the one with a square on the left side) on the `SmartPad`. Application pauses, breaks are activated.
- Close the application:
    - Sunrise OS < 1.15: 
       pressing the 'X' button on the top right 
    - Sunrise OS >= 1.15:
       pressing the reset button from the Applications drop-down menu
- Shutdown any ROS node or `roscore` running (you can also leave roscore running, the robot will reconnect without any problem the next time).

There should not be any error, if you can reproduce some steps that always give you an exception please let me know.

BrunoScaglioni, 2019-07-02:
Hi Salvo,

the steps you mentioned are exactly the same we do when starting/stopping the application.
i didn't realize that the BindException is thrown after the application has previously stopped with an error but this makes sense, as dispose() is not called correctly. On our platform sometimes the application does not stops within 15 seconds and this triggers an error in the scheduler that terminates it, that is why we often have this issue. I don't know if there is a way to extend this time or prevent the scheduler from stopping the application.

SalvoVirga, 2019-07-02:
It is strange that you have that often, to us it happens only when somehow we mess up something. But I don't have yet a good way to reproduce the error at will.
I believe those 15 seconds are set by KUKA, whenever you stop the application, it will be forcefully stopped if it doesn't exit cleanly after 15 seconds.

In any case, however, that `dispose()` method is an override of the one in the base class `RoboticsAPIApplication` and it should just be called anytime the application is 'disposed', so it should always be called automatically, just like` initialize()` and `run()`.
Since it seems that this is not the case, we have an explicit call [in the catch block of the main application loop](https://github.com/IFL-CAMP/iiwa_stack/blob/master/iiwa_ros_java/src/de/tum/in/camp/kuka/ros/app/ROSBaseApplication.java#L363). It might worth a try to move this call to the `finally` block actually, that should be its proper spot.

But again, `dispose()` should be always called internally by KUKA when the application is closed, in any case, if it is not it is a problem on their side.

jacknlliu, 2019-08-20:
The issue after terminating the previous one.

jacknlliu, 2019-08-24:
@SalvoVirga  Is there any relationship with https://github.com/rosjava/rosjava_core/pull/291 ?

iwt-vsp, 2019-09-02:
Hello,

I am also encountered with similar issues after I updated iiwa_ros_java in Sunrise Workbench. I have followed the steps provided by @SalvoVirga in the above discussion. Also, I have followed the issue #190 and pinged robot and ROS master system to each other and it is successful. However, I am not able to remove the error. 

When I connect it for the first time I get the following error.
[image]

When I connect for the second time and onwards.
[image]

Is there any working solution? 

Thank you.

AlessandroPiz, 2024-07-29:
I personally found that RosSmartServo is correctly closed if and only if i close it before stopping the ROS master