IFL-CAMP/iiwa_stack #184: Cannot Install ROSSmartServo

RagPrE, 2019-06-17:
I followed the instructions of setting up the sunrise cabinet, but I still cannot find the ROSSmartServo application on the pad and I got a compiler error on the sunrise workbench. I do not know where I miss the step. Could anyone give me some instructions on how to resolve this issue?

The followings are the screenshots of my current sunrise workbench.
  **compiler error: & workbrench**
[image]

**my RoboticsAPI.data.xml**
[image]

**software options on station setup.cat**
[image]

SalvoVirga, 2019-06-17:
I see a red exclamation mark on the Sunrise Project, is there any item in the Error panel on Sunrise Workbench?

RagPrE, 2019-06-17:
> I see a red exclamation mark on the Sunrise Project, is there any item in the Error panel on Sunrise Workbench?

INo, I can install this project without any problems, but the ROSSmartServo is still missing on the panel.

SalvoVirga, 2019-06-17:
That is strange, try to refresh the project (Right Click on it -> Refresh) and see it that exclamation mark goes away. Maybe try to reinstall the project and then sync it, or make a new project from scratch, install that, sync that, then apply the modification needed for iiwa_stack and resync.

RagPrE, 2019-06-17:
> That is strange, try to refresh the project (Right Click on it -> Refresh) and see it that exclamation mark goes away. Maybe try to reinstall the project and then sync it, or make a new project from scratch, install that, sync that, then apply the modification needed for iiwa_stack and resync.

Here, you can see the error message went away, but I still cannot found the application on the panel.
[image]

One additional warning:
[image]

SalvoVirga, 2019-06-17:
You synced the project after installing it, right?
You can also try to create a dummy RoboticApplication, empty, unrelated to iiwa_stack and sync again, that should appear.

If all of that doesn't work, please try to create a new project and add the demo RoboticApplications that ship with Sunrise Workbench, sync the project and those application have to appear.

RagPrE, 2019-06-17:
> You synced the project after installing it, right?
> You can also try to create a dummy RoboticApplication, empty, unrelated to iiwa_stack and sync again, that should appear.
>
> If all of that doesn't [...]

RagPrE, 2019-06-17:
> You synced the project after installing it, right?
> You can also try to create a dummy RoboticApplication, empty, unrelated to iiwa_stack and sync again, that should appear.
>
> If all of that doesn't [...]

I forgot to synced. Maybe that is the problem. I will try that later.

RagPrE, 2019-06-17:
> You synced the project after installing it, right?
> You can also try to create a dummy RoboticApplication, empty, unrelated to iiwa_stack and sync again, that should appear.
>
> If all of that doesn't [...]

Thanks for your help.

closed 2019-06-17 by RagPrE

BossaMelon, 2019-06-26:
> Thanks for your help.

Can Sunrise Workbench support Chinese Language?