IFL-CAMP/iiwa_stack #185: Is the helper class iiwaRos deprecated?

isabellahuang, 2019-06-19:
C++ examples in the wiki refer to iiwaRos, but it seems like this helper class doesn't exist anymore. What should we use instead?

dkilanga3, 2019-06-21:
I have been wondering the same thing for the last 2 days

SalvoVirga, 2019-06-21:
Well, yes, those examples are very very old, they were written around 2 years ago and the APIs changes a lot from then.
You could have a look at the headers in the `iiwa_ros` package, the interface is pretty straight forward.

Now there are different classes for each state topic, command topic and service.

Examples:

- Read the robot Cartesian pose:

```cpp
#include <iiwa_ros/state/cartesian_pose.hpp>

int main() {

  iiwa_ros::state::CartesianPose pose_state;
  pose_state.init("iiwa") // That is the namespace under which the topic lives.

  while(true) {
    auto pose = pose_state.getPose();
  [...]
  }

}
```

You can also initialize the object passing a callback function that will be called every time a new message arrives.

- Send a Cartesian pose as command to the robot:

```cpp
#include <iiwa_ros/command/cartesian_pose.hpp>
#include <geometry_msgs/PoseStamped.h>

int main() {

  iiwa_ros::command::CartesianPose pose_command;
  pose_command.init("iiwa") // That is the namespace under which the topic lives.

  geometry_msgs::PoseStamped commanded_pose;
  commanded_poses = [...]

  pose_command.setPose(command_pose);
}
```

You can also give a callback function to setPose, that function will be called as soon as the commanded motion is complete.

The exact same interface is provided for all the other state and command topics.
For services is also very similar: create the object, initialize it, call their function to use the service.

Please check the headers for more info, they should all be documented.

dkilanga3, 2019-06-21:
Thank You for clarification! It is really appreciated

dkilanga3, 2019-06-21:
> Well, yes, those examples are very very old, they were written around 2 years ago and the APIs changes a lot from then.
> You could have a look at the headers in the `iiwa_ros` package, the interface is pretty straight forward.
> 
> Now there are different classes for each state topic, command topic and service.
> 
> Examples:
> 
> * Read the robot Cartesian pose:
> 
> ```c++
> #include <iiwa_ros/state/cartesian_pose.hpp>
> 
> int main() {
> 
>   iiwa_ros::state::CartesianPose pose_state;
>   pose_state.init("iiwa") // That is the namespace under which the topic lives.
> 
>   while(true) {
>     auto pose = pose_state.getPose();
>   [...]
>   }
> 
> }
> ```
> 
> You can also initialize the object passing a callback function that will be called every time a new message arrives.
> 
> * Send a Cartesian pose as command to the robot:
> 
> ```c++
> #include <iiwa_ros/command/cartesian_pose.hpp>
> #include <geometry_msgs/PoseStamped.h>
> 
> int main() {
> 
>   iiwa_ros::command::CartesianPose pose_command;
>   pose_command.init("iiwa") // That is the namespace under which the topic lives.
> 
>   geometry_msgs::PoseStamped commanded_pose;
>   commanded_poses = [...]
> 
>   pose_command.setPose(command_pose);
> }
> ```
> 
> You can also give a callback function to setPose, that function will be called as soon as the commanded motion is complete.
> 
> The exact same interface is provided for all the other state and command topics.
> For services is also very similar: create the object, initialize it, call their function to use the service.
> 
> Please check the headers for more info, they should all be documented.

Do we do have to create node with the above code or can we just run it in a C++ executable file?

SalvoVirga, 2019-06-25:
> Do we do have to create node with the above code or can we just run it in a C++ executable file?

Sorry but this question is not clear to me. You can use the C++ classes provided by `iiwa_ros` on any other C++ code. You can make a ROS node (which is just a C++ executable) that uses that code, a library, whatever.

SalvoVirga, 2019-06-28:
Can this be closed?

SalvoVirga, 2019-08-19:
I'll close this, please feel free to reopen it if still needed.

closed 2019-08-19 by SalvoVirga