IFL-CAMP/iiwa_stack #186: Cannot control CartesianPose

isabellahuang, 2019-06-20:
I'm currently having trouble commanding the CartesianPose of the robot. I am publishing a geometry_msgs/PoseStamped message to /iiwa/command/CartesianPose, but the robot is not moving. I see a potential solution on how to fix this in the [message definition file](https://github.com/IFL-CAMP/iiwa_stack/blob/master/iiwa_msgs/msg/CartesianPose.msg) but I'm unclear on which "value to copy" and which parameters to set to -1. In fact, I can't even find this "Cartesian Position tab" on the SmartPad. I would really appreciate some additional clarification! 

Furthermore, my attempts to publish the CartesianPose command leads to this error on the SmartPad. 
[image]

I'm pretty confident my numbers are within acceptable range. But I've also published the exact values from the current CartesianPose state as a sanity check (ie. no movement) but I get the same error. Thank you for reading!

SalvoVirga, 2019-06-21:
> I'm pretty confident my numbers are within acceptable range.

This can be, but the error from the SmartPad seems clear: the controller computer some path to reach your commanded destination and some configuration in joint space during that path would violate the robot limits. That could be possible, even if the cartesian pose is reachable, maybe it is not from the current robot configuration. Or maybe your commanded position is correct, but not the orientation.

- Did you try to command the same pose from another initial joint configuration?
- More importantly, did you try to command any other pose to the robot? Easy toy example: Take the current robot flange Cartesian Position, modify one components of its position by 1cm or so, send the modified pose (with the same orientation of the initial one). Does the robot move?
You should try to break down your errors to toy examples, if the robot moves with simple commanded poses (e.g. poses that are very close to the current one and therefore very likely to be correct, unless you are already very close to a joint limit), then you are probably not sending something correct. Or, you are very unlucky and the robot simply cannot achieve that end-effector pose from the current configuration.

isabellahuang, 2019-06-21:
Thanks for the response. I have been trying these simple examples, but as I
mentioned even when I command it to stay in the same orientation I get the
out of bounds error. I will definitely keep trying some other examples and
be extra mindful of the orientation.

However, I would still love some additional clarification on what the
comments in the CartesianPose msg file means. Ie, what parameters to set to
-1? Thanks!

exo-core, 2019-07-23:
When you want to ignore the redundancy data in a `iiwa_msgs/CartesianPose` you need to set `status` and  `turn` to `-1`:

```
~$ rosmsg show iiwa_msgs/CartesianPose 
geometry_msgs/PoseStamped poseStamped
  std_msgs/Header header
    uint32 seq
    time stamp
    string frame_id
  geometry_msgs/Pose pose
    geometry_msgs/Point position
      float64 x
      float64 y
      float64 z
    geometry_msgs/Quaternion orientation
      float64 x
      float64 y
      float64 z
      float64 w
iiwa_msgs/RedundancyInformation redundancy
  float64 e1
  int32 status // set this to -1
  int32 turn   // and this as well
```

isabellahuang, 2019-08-12:
> When you want to ignore the redundancy data in a `iiwa_msgs/CartesianPose` you need to set `status` and `turn` to `-1`:
> 
> ```
> ~$ rosmsg show iiwa_msgs/CartesianPose 
> geometry_msgs/PoseStamped poseStamped
>   std_msgs/Header header
>     uint32 seq
>     time stamp
>     string frame_id
>   geometry_msgs/Pose pose
>     geometry_msgs/Point position
>       float64 x
>       float64 y
>       float64 z
>     geometry_msgs/Quaternion orientation
>       float64 x
>       float64 y
>       float64 z
>       float64 w
> iiwa_msgs/RedundancyInformation redundancy
>   float64 e1
>   int32 status // set this to -1
>   int32 turn   // and this as well
> ```
> 
> You can also take a look at this [spline motion example](https://github.com/exo-core/iiwa_stack_examples/blob/development/iiwa_actions/src/SplineMotionDemo.cpp).

How should I set it if iiwa/command/CartesianPose only accepts the geometry_msgs/PoseStamped part rather than the entire iiwa_msgs/CartesianPose message?

Deastan, 2019-08-16:
I have exactly the same error! Did you find a solution ?

isabellahuang, 2019-08-16:
@Deastan Unfortunately I have not been able to fix the problem! Same with me-- commands only work in joint space. I've been using TRAC-IK as a workaround. 

Also, what's interesting is that when I try to control the Cartesian pose linearly (through iiwa/command/CartesianPoseLin) my robot jerks a little violently but is immediately stopped by the speed limit detection on the KUKA. I was excited because at least it moved. So I tried to set the linear Cartesian velocity to be slower through the services (see my issue #207) but to no avail. I am however able to control the joint velocity through these services. 

I'd like to see what happens when you try to call the CartesianPoseLin command instead of the CartesianPose command. Could you let me know?

Deastan, 2019-08-16:
Should I run Moveit before to command a desire pose ? 
I have no longer the error, but the robot doesn't move... I think it is not receiving the command..

isabellahuang, 2019-08-16:
I'm not currently using Moveit and I don't think it's necessary in theory. Yeah, that error shows up when you publish an /iiwa_msgs/CartesianPose message instead of a geometry_msgs/PoseStamped to the /iiwa/command/CartesianPose or /iiwa/command/CartesianPoseLin topics. 

It's a little bit confusing because reading the state /iiwa/state/CartesianPose returns a iiwa_msgs/CartesianPose message but commanding it requires a PoseStamped message. You could also follow Salvo's example code for using the helper functions that he typed up for me in #185 . 

When you publish to the topic /iiwa/command/CartesianPoseLin are you getting any errors?

Deastan, 2019-08-16:
I have no error/no warning.

I have the git from March... because I change and adapt for gazebo because I was building an RL environment... I have added this msg : CartesianPose because it wasn't on iiwa_msgs. Maybe something else is missing. :/

isabellahuang, 2019-08-16:
Sorry, what do you mean by adding msg : CartesianPose? I don't quite
understand.

I know a lot of changes were pushed in June so it could be worth trying to
send a basic CartesianPose command with the updated code if that's feasible
for you. Though I'm updated with the master branch and still having errors.

I also have a suspicion that attaching tools to the arm may affect whether
commanding the CartesianPose works. I saw #159 has the same error we have.
I personally have attached a tool but have yet to remove it and try the
Cartesian commands again. Are you maybe also using a tool?

Deastan, 2019-08-16:
So I found out ! I will answer you soon.

isabellahuang, 2019-08-16:
Oh, that's great that it works! I'm still trying to make very small incremental changes to my current CartesianPose but the same error persists...

Deastan, 2019-08-16:
If you check the joints limits space on the pad ? Try first to reach x =0,
y = 0 and a = 2 meter. To see if the error message on the pad change to
cannot compute the inverse kinematic. What is happening ? Try using my
python scrip if it is simple to set or rqt. Be careful with the orientation.

davidpaulius, 2022-08-30:
This may be really late, but I had a similar issue when using the CartesianPose message type with the CartesianPose command. Instead, I had to use the PoseStamped message type to get the robot to move. Hope this helps anyone who is stumped on this issue!

shrutic12, 2023-02-23:
Hey all, I have a strange error, where publishing on CartesianPose or CartesianPoseLin does not move the robot, however, when I go to moveit path planning and manually move the robot along the same direction, it works. So, in this case I am certain the joint limits have not been reached. But it seems like the publishing to the topic gives no result. Any ideas what I could try? I also have no errors.

exo-core, 2023-02-27:
> Hey all, I have a strange error, where publishing on CartesianPose or CartesianPoseLin does not move the robot, however, when I go to moveit path planning and manually move the robot along the same [...]

Did you forget about the namespace? It should be `/iiwa/CartesianPose` or `/iiwa/CartesianPoseLin`, if I remember correctly...

shrutic12, 2023-02-27:
Hi, No, I did not forget the namespace, it was just to keep the post short. I am using a KUKA LBR Med robot with arealsense camera attached. I am using a trained neural network to detect some markers and I would like to have a visual servoing type application where the robot is able to track the markers. Currently, I compute the pose in the camera frame, which is then transformed to the base frame of the robot and I am sending the pose commands to /iiwa/CartesianPose. But the tracking becomes unstable and the robot stops moving. Any ideas/suggestions would be great!