IFL-CAMP/iiwa_stack #188: Multiple iiwa IP/change iiwa IP for X66 port

emielke12, 2019-06-25:
With the latest merge #178, we no longer are setting the KONI port IP address, and therefore are essentially stuck using the 172.31.1.147 address for the X66 port. Are there any ways to get around using this IP (I remember previously it was fatal to change this IP for any reason)? How would one go about trying to run multiple iiwa robots?

SalvoVirga, 2019-06-25:
You can change that IP from Sunrise Workbench, you just have to be careful to not change it to some specific subnets that are forbidden. This should all be documented in the KUKA documentation.

Using iiwa_stack with multiple robots should be definitely possible!

emielke12, 2019-07-22:
Have you (or anyone) had any issues changing to a different IP? I tried to change onto a 192.168.XXX.XXX subnet, changed the IP in the configuration file, and set the address range for KLI to 192.*.*.* However, when I attempt to install the program, I cannot see the IP set (can't even ping or anything, and I end up having to use the recovery stick. 

Contacted KUKA, but they have been less than helpful. So far as I am aware I am doing everything according to the sunrise cabinet manuals. Is there any trick to getting this to work?

exo-core, 2019-07-23:
We have changed the cabinet IP to `192.168.1.xxx` and did not encounter any problems so far. Not sure if we also changed the address range for KLI too, though.

This is what you need to do:

1. Set your laptop's IP to an address in the same range as the one of the Sunrise cabinet (by default somewhere in `172.31.1.xxx`)
2. If necessary download current project from the controller to Sunrise workbench
3. Change the IP in the `StationSetup.conf`
4. Select install. Sunrise workbench will try to ping the workstation at the newly configured IP and report that it could not find a controller. Double click on the current IP field in that dialog and change it to the cabinet current IP (default is `172.31.1.147`). The background color should switch to green and you can run the installation.
5. Restart the controller.
6. Change the IP of your laptop to lie in the cabinets new IP range (`192.168.1.xxx`). You should now be able to reach the cabinet under it's new IP.
7.  Adjust the ROS master IP in the `iiwa_stack` configuration and synchronize it with the controller.

I wrote everything down from the top of my head, so there might not be 100% accurate...

exo-core, 2019-07-23:
One more remark, that should also be written in the Wiki: if I remember correctly the IPs for KONI and X66 needed to be in different address ranges, so we assigned X66 to `192.168.1.xxx` and KONI to `192.168.2.xxx`.

emielke12, 2019-08-07:
Pinged KUKA about this. The instructions in the KUKA manual appear to be correct, as well as the process you've outlined, however it didn't work for my system. I'm going to close this as not even KUKA has suggestions beyond not using the subnet.

closed 2019-08-07 by emielke12