IFL-CAMP/iiwa_stack #190: KONI-Port

kellerjo, 2019-06-26:
Hey there,
First of all: Congratz to the new version!
The SmartServo still worked for us after switching to the new master branch. But after removing the KONI-Port cable and switching back to the original settings (-os RTOS), the ROSSmartServo application crashes immediately with a NullPointerException followed by a big wall of errors.  

[image]

note: we are not using the default ip adress for the robot

Did anyone run into a similar problem?

SalvoVirga, 2019-06-27:
Very strange, especially cause the error says it is trying to connect to `0.0.0.0`, so it looks like somewhere you have a error in your network setup. Please be sure that the IP on the X66 is correct, that the settings on the ProcessData tab are correct (no settings there should make the application use the default IP 172.31.1.147) and that you can ping the robot from the ROS master machine.

kellerjo, 2019-07-01:
thanks for your reply!
The robot can be pinged and the ip is correct in the ProessData. Also setting the IP back to default (172.31.1.147) yields the same error.
After plugging in the KONI-Port again and setting it to WIN, it works fine again

iwt-vsp, 2019-08-30:
@kellerjo Can you please tell me what do you mean by WIN?