IFL-CAMP/iiwa_stack #191: About connectivity Package

BossaMelon, 2019-06-26:
I've recently ask KUKA for the Connectivity and Servoing Packages. And they reply me"do you mean servoing and FRI". Seems that Connectivity Package is already a legacy  name? I've read the issues and don't really clear. Is Servoing already enough for the iiwa_stack if I don't need 1kHZ command rate. And for the connection between the Robot and Ros Machine we just use the X66 Port? Not KONI?
Thanks guys!

SalvoVirga, 2019-06-27:
> I've recently ask KUKA for the Connectivity and Servoing Packages. And they reply me"do you mean servoing and FRI". Seems that Connectivity Package is already a legacy name?

Yes, Servoing and FRI are what was called the "Connectivity package".

> Is Servoing already enough for the iiwa_stack？

At this point in time, yes, Servoing is enough. There is a fork of iiwa_stack that uses FRI tho, if you need that.

> For the Java code in src folder do I need to modify the content like
> "import com.kuka.connectivity.motionModel.smartServo " to something else? Because connectivity is already an old name of the package？

No, the code in the master branch should work with the latest versions of Sunrise.

BossaMelon, 2019-06-27:
Thanks a lot for your kind reply!

chenhaowen01, 2019-06-28:
I would like to kown where could I get the document of the "Connectivity package". I only have two manul from KUKA, "Operating Instructions for End Users" and "Operating and Programming Instructions for System Integrators", neither of which mentioned about "Connectivity package".

SalvoVirga, 2019-06-28:
The Connectivity/Servoing package comes on its own as a plugin for Sunrise Workbench. You should contact KUKA support to get it if you don't have it, they will also provide the documentation.

chenhaowen01, 2019-06-29:
OK, thank you.

BossaMelon, 2019-08-26:
Hello Salvo,

I‘ve just ask kuka about the Package. They say that every version of sunrise workbench can install the servoing package(newest version is 1.16 for now), and just need to be paid.

> The Connectivity package is not available for every version of Sunrise.OS, so far it was available for odd versions (1.5, 1.7, 1.9, 1.11, 1.13, 1.15).

You mean here the package are free to get for odd versions?

closed 2020-01-01 by BossaMelon