IFL-CAMP/iiwa_stack #192: Unable to import ServoMotion

madhukartlr, 2019-06-27:
Hi everyone,

I'm trying to setup the Sunrise Project with iiwa_stack. After synchronizing the project, I got the following errors in de.tum.in.camp.kuka.ros and de.tum.in.camp.kuka.ros :

import com.kuka.connectivity.motionModel.smartServo.ServoMotion
import com.kuka.connectivity.motionModel.smartServo.SmartServo
import com.kuka.connectivity.motionModel.smartServoLIN.SmartServoLIN
import com.kuka.task.ITaskLogger

I'm unable to import them. Any help is appreciated.
Thanks

SalvoVirga, 2019-06-27:
> PS: I have already installed the Sunrise.Connectivity using com.kuka.connectivity.industrial.repository-1.7.0.ConnV17_2 but I am still having issues with importing Servo Motion controls.

I'm not sure what that is, but the errors mean that you did not installa the `Servoing` package from KUKA. Please install that one, in the `Software` tab of the `StationSetup` of your project you should see and activate the `SmartServo` and `SmartServoLIN` packages

madhukartlr, 2019-07-01:
In the software tab of StationSetup of my project, I could see that **Smart Servo Motion Extension**, **Smart Servo Linear Motion Extension** and **Direct Servo Motion Extension** are installed and activated already. Still the problem persists.

SalvoVirga, 2019-07-01:
Could you post a screenshot of your StationSetup file and the errors you have in the project?
Please be sure to refresh the project (F5 or Right Click on the project -> Refresh) before.

closed 2019-07-01 by madhukartlr