IFL-CAMP/iiwa_stack #193: iiwa in Gazebo

dkilanga3, 2019-06-28:
Does the gazebo simulation work with all the functionalities offered in this package?
I am publishing joints command on the topic iiwa/command/JointPosition, and when I echo that topic, I am able to see that my commands are really being sent. However, the robot in gazebo does not move, and when I echo the iiwa/state/JointPosition, I am told that nothing is being received.
Do you have any idea of what is going on and how I can solve the issue?

SalvoVirga, 2019-06-28:
If you just spawn the robot on Gazebo, well, you only have the robot in Gazebo! You should be able to use it with MoveIt!. I don't really use Gazebo, so I don't have specific suggestions.
If would be nice if someone that uses it would add an entry for the wiki.
The topics you mention live on a real robot's cabinet, so they are not available in simulation.

That said, with the latest release PR #151 introduced the simulation of the Sunrise Controller for Gazebo. That means that if you run the launch `iiwa_gazebo_with_sunrise.launch`. You will get the robot on Gazebo and a small node that simulates the Sunrise Controller and creates some of the topics available for the real robot. You might want to have a look at that.

Deastan, 2019-08-16:
Hello, did you solved your problem ? 
If not, the topic is "/iiwa/PositionJointInterface_j4_controller/command" for the joint 4! And then set the value with -1.57 for instance and you should have 
[image]

Don't forgot to change in the iiwa_gazebo.launch the parameter: trajectory to false!

The one that you was using is to command the real robot using sunrise... 

Hope it can help you!

Deastan, 2019-08-16:
@SalvoVirga With the second part of your message, you mean that we can simulate the inverse kinematics of the kuka in the simulation removing MoveIt?

SalvoVirga, 2019-08-19:
@Deastan yes, using the code introduced in #151, have a look at that.

closed 2019-08-19 by SalvoVirga