IFL-CAMP/iiwa_stack #196: Sunrise cabinet and ros master machine not working

bhuris, 2019-07-08:
I am following the startup procedure from author ‘s suggestion on issue 138 
Now I can run roscore and rostopic on rosmaster without error 
And on the sunrise controller remote only shown waiting for ros master to connect 
[image]
[image]
Now i’m very confusing because i could ping the rosnode without problem
The setting on the robot are shown in pic below
[image]
Please give me a recommendations 
I have tried to restart both of ros and reinstalling ros kinetic

SalvoVirga, 2019-07-08:
What are the errors in the SmartPad? Do you get them all the time you start the application?

bhuris, 2019-07-09:
Run roscore on master machine and nothing happen on smartpad
[image]

SalvoVirga, 2019-07-09:
Just to be sure, the ROS machine and the cabinet are connected using the X66 port, right?

bhuris, 2019-07-09:
Ans- the rosmaster and kuka connect via x66 and ping success 
Only get error while restart Rossmartservo app on smartpad 
[image]
[image]

But sunrise os that sync in it working fine ,before this i already setup with another rosmaster on vmware and it works then I setup on this pc (i re-sync sunrise os again )

matthiashh, 2019-07-09:
The restart for me looks like #183 to me. I would treat it as a separate issues. Just remember that the current solution is to restart the cabinet: https://github.com/IFL-CAMP/iiwa_stack/issues/183#issuecomment-501593500

I had issues with `melodic` when I tried to use `ROS_HOSTNAME` instead of IPs, but you seem to use IPs and `kinetic`.
One thing you can try is if you can ping your rosmaster from the cabinet. Either with plugging in a screen at start up, a USB keyboard to the smartpad or RDP into your cabinets IP with the default cabinet password to get the same screen as the smartpad.

bhuris, 2019-07-09:
@matthiashh thanks for suggesting, i can not ping from kuka cabinet !
[image]
And the ip address above on pic isnt match to ip setting in ‘process data’
‘

matthiashh, 2019-07-09:
You seem to have found a menu I haven't seen yet.
I would have tried using the underlying windows that could supply a normal command line interface. Either from the smartpad or the cabinet by plugging in a monitor at startup.

Another error I did once was to not specify the correct IP address in the process data configuration and just changing the first address and not the second
```
<processData dataType="java.lang.String" defaultValue="172.31.1.150" displayName="ROS Master IP" editableOnHmi="true" id="master_ip" value="172.31.1.150"/>
```

bhuris, 2019-07-09:
@matthiashh recently my KUKA sales representative told me this is not suitable for check ping to ros master
this screen is shown ip of my smartpad controller
so pinging is need to be test on command line interface 
but from now still have problem to connectivity with ros interface
I don't know what the error happening ,cause they didn't shown any error message at all
I am already check this ping from rosmaster to sunrise,check source in ros setting ros ip-port,following installing procedure from wiki,check sequence of startup-shutdown of rossmartservo on smartpad,run roscore and [echo rostopic](https://github.com/IFL-CAMP/iiwa_stack/wiki/readrobotstate) successfully without an error.
then ros topic echo result is "warning topic...does not appear to be published yet"
(pic above)

matthiashh, 2019-07-09:
>  so pinging is need to be test on command line interface

Well that you can easily test.

Maybe I would give it a try to use the `KONI` network interface. The current version I am running here is the old `development` branch right before it was merged into master. So it should work with `X66` (and it did), but recently I noticed that I can not reinstall the controller when I am connected to `X66` only and had to switch to `KONI` for that (which is a separate issue). Maybe it's worth a shot.

Did you try if you can connect to your ros master from other computers in the network?

bhuris, 2019-07-10:
I can ping ros master from sunrise cabinet !
[image]
[image]
[image]
I’m using HP proliant dl380p gen8 server as my rosmaster machine

robotlinker, 2019-07-28:
you may need to press the play button on the control panel, just give a try

SalvoVirga, 2019-08-19:
any updates on this?

BossaMelon, 2019-08-26:
Hello,

just would like to know how can I use the command line in the Smartpad?

matthiashh, 2019-08-26:
> Hello,
> 
> just would like to know how can I use the command line in the Smartpad?

There are several ways. One of them is to plug in a keyboard to the USB port on the back and press the Windows key and open a terminal. Another one would be to RDP connect to your Cabinet and do the same from your computer.

BossaMelon, 2019-08-29:
> There are several ways. One of them is to plug in a keyboard to the USB port on the back and press the Windows key and open a terminal. Another one would be to RDP connect to your Cabinet and do the [...]

Hello matthiashh，

it works, really appreciate your help!