IFL-CAMP/iiwa_stack #198: tips about merging the master branch to Sunrise 1.7

jacknlliu, 2019-07-26:
Hi, for many reasons, we can't use the latest version of this repo. I hope to continue to maintain the Sunrise 1.7 version. Could you give me some tips about how to modify the master branch to Sunrise 1.7?

SalvoVirga, 2019-07-26:
It shouldn't be too hard.
Try to use the current (in master) Java code in Sunrise Workbench, using the APIs from Sunrise 1.7. You will most likely have some import error. Delete the imports that give you those errors, then you are going to have errors for missing references in the code.
Just over on top of the lines whose references are missing and you should automatically get a suggestion on what to import to fix the problem.

The only thing you might need extra is the file [xml-apis-1.0.b2.jar](https://github.com/IFL-CAMP/iiwa_stack/blob/ae3ba2792048f73122edfe1abb5a3e6b25540c45/iiwa_ros_java/ROSJavaLib/xml-apis-1.0.b2.jar) that you can find in some very old commit like the one linked.

If you encounter particular errors let me know.

GuancongLuo, 2019-08-08:
I meet the same problem. 
And I can import some package
`import com.kuka.connectivity.motionModel.smartServo.ServoMotion;`
`import com.kuka.connectivity.motionModel.smartServo.SmartServo;`
`import com.kuka.connectivity.motionModel.smartServoLIN.SmartServoLIN;`
`import com.kuka.task.ITaskLogger;`

Actually, I have the 'conectivity' package, but the package name is different with the code.

jacknlliu, 2019-08-08:
@GuancongLuo We have updated the Sunrise OS to 1.11 using another backup hard disk, so I may not maintain the 1.7 branch recently. Good luck anyway！

SalvoVirga, 2019-08-19:
@GuancongLuo Did you try to remove the imports that trigger the errors? Workbench should then mark the symbols with a missing import and suggest which import to add.
If that doesn't help, just explore the KUKA API's a bit, you should be able to find the right imports quite easily. You will also need to change the `TaskLogger` import btw, that change with some more modern version.

Let me know if you solved it, else I could try to install a Sunrise 1.7 on a VM

jacknlliu, 2019-08-20:
@GuancongLuo I have committed a branch for Sunrise 1.7 [here](https://github.com/jacknlliu/iiwa_stack/tree/update-v1.7-from-upstream).

GuancongLuo, 2019-09-11:
@jacknlliu Thank you for your branch. I successfully installed **ROSSmartServo** application. 
@SalvoVirga https://github.com/IFL-CAMP/iiwa_stack/issues/183#issuecomment-502653195
But iiwa 14 can't connect successfully with my ROS. I check the ROS IP and sunrise IP, this is right. I don't know how to solve this problem. Could you get me some advance?

[image]
ROS core and Ethernet address

[image]

Sunrise WorkBeach RoboticsAPI.data.XMl

[image]
Always waiting. And when I stop it, I get the error.

I need your help and thanks again.

jacknlliu, 2019-09-11:
@GuancongLuo  make sure that you are using the **X66** port, and could you ping the robot ip from the PC?

GuancongLuo, 2019-09-12:
Yes, I can ping the robot IP from PC. And I connected the PC with **X66** port this morning. I started **roscore** first and then started **ROSSmartServo**. It is not working.

[image]
[image]

Still, have the same problem.

[image]

Is this a **configuration** issue for **Sunrise WorkBeach**?

jacknlliu, 2019-09-13:
It's nice to hear that it works for you.