IFL-CAMP/iiwa_stack #200: How to convert iiwa_msgs::CartesianPose to geometry_msgs::postStamped

Tigerzuo, 2019-07-26:
I have a function that to move robot arm by some given value, however getPose only return iiwa_msgs but setPose() takes in a geometry_msg and I am not sure how to make those compatible. Thanks!

`void movearm(int x, int y){
    auto mypose = pose_state.getPose(); //robot current pose
    mypose = mypose.pose.position.x + x;
    mypose = mypose.pose.position.y + y;
    pose_command.setPose(mypose);

  }`

exo-core, 2019-07-26:
If you take a look at `iiwa_msgs::CartesianPose` you will see that it actually consists of a `geometry_msgs::postStamped`:
```
geometry_msgs/PoseStamped poseStamped
  std_msgs/Header header
    uint32 seq
    time stamp
    string frame_id
  geometry_msgs/Pose pose
    geometry_msgs/Point position
      float64 x
      float64 y
      float64 z
    geometry_msgs/Quaternion orientation
      float64 x
      float64 y
      float64 z
      float64 w
iiwa_msgs/RedundancyInformation redundancy
  float64 e1
  int32 status
  int32 turn
```

Tigerzuo, 2019-07-26:
sweet, thanks!

closed 2019-07-26 by Tigerzuo