IFL-CAMP/iiwa_stack #207: How can I set the CartesianPose command velocity?

isabellahuang, 2019-08-12:
I am currently using the /iiwa/command/CartesianPoseLin topic to publish my CartesianPose commands to the robot. This is because /iiwa/command/CartesianPose doesn't work for me. However, the robot moves very fast and the KUKA is unable to execute the full motion due to the "speed limit detected" showing up on the SmartPad and stopping the execution. 

Is there a way to change the Cartesian velocity to make the robot go slower? Perhaps I need to set the ControlMode using the configuresmartservo service. I would really appreciate a step by step answer as well! Thank you.

exo-core, 2019-08-13:
You can use the `/iiwa/configuration/setSmartServoLinLimits` service to configure the Cartesian velocity limits.

isabellahuang, 2019-08-14:
Does this just change the allowable velocity on the Kuka system? I would
like to actually move the arm slower, not just change the limits. Thanks!

isabellahuang, 2019-08-14:
I used the setSmartServoLinLimits service to set very low allowable velocities, but it doesn't seem to change anything unfortunately. Would appreciate any further tips.