IFL-CAMP/iiwa_stack #209: relationship about cartesian wrench data and tool

jacknlliu, 2019-08-13:
If I have set the tool as the [wiki](https://github.com/IFL-CAMP/iiwa_stack/wiki/attachtool), does the measured wrench data from topic `/iiwa/state/CartesianWrench` include the load of the tool?

SalvoVirga, 2019-08-19:
/iiwa/state/CartesianWrench will output the [external force/torque measured by the robot](https://github.com/IFL-CAMP/iiwa_stack/blob/master/iiwa_ros_java/src/de/tum/in/camp/kuka/ros/MessageGenerator.java#L146) according to the used reference frame.
If you use a tool and define its tip as the reference frame, you will get any external force/torque applied at that point. If you use the robot's flange, the signal should include the tool load.

jacknlliu, 2019-08-20:
I set the load data(0.13 KG) for the tool in Sunrise workbench using the SmartHMI, but set the `tool_link_ee` same as the flange.

I find that, before calibrating the load using SmartHMI, the data from topic `/iiwa/state/CartesianWrench` is:
```
header: 
  seq: 183092
  stamp: 
    secs: 1356996939
    nsecs: 493000000
  frame_id: "tool_link_ee"
wrench: 
  force: 
    x: -1.12026890559
    y: -0.711754420191
    z: 1.392632514
  torque: 
    x: -0.170390946122
    y: -0.194638857823
    z: -0.195380094373
inaccuracy: 
  force: 
    x: 2.97890978144
    y: 2.33205263901
    z: 2.50024415521
  torque: 
    x: 0.830721101649
    y: 0.952830475078
    z: 0.577498320064
``` 

after calibrating the load

```
header: 
  seq: 20332
  stamp: 
    secs: 1356995215
    nsecs: 936000000
  frame_id: "tool_link_ee"
wrench: 
  force: 
    x: -0.521026506566
    y: -0.793812966469
    z: 0.480188497791
  torque: 
    x: 0.00826004237117
    y: -0.411171071362
    z: -0.23004151063
inaccuracy: 
  force: 
    x: 2.97959041273
    y: 2.33224845974
    z: 2.50038630278
  torque: 
    x: 0.830703980632
    y: 0.952918812057
    z: 0.57749908983
```

It seems that the robot controller minius the load data, and use this measured data to transform to the specified frame.

SalvoVirga, 2019-08-21:
Using the KUKA APIs I guess they take into account the tool load calibration when measuring external forces. If instead of using `tool_link_ee`, you would use `iiwa_link_ee` as the reference frame (still, having the tool attached and calibrated), the signal should contain also the force/torque introduced by the tool. So this behavior is to be expected.

You should look into the Java code, we have two variables `toolFrame` and `endpointFrame` that are used to configure the tool and the reference frame for the motions. At this points the output of the robot Cartesian state (pose, wrench), depend on the reference frame used for the motion command (i.e. if you command in `tool_link_ee`, you cannot get the state wrt `iiwa_link_ee`). If you need something like that, some work would be needed on the Java side.