IFL-CAMP/iiwa_stack #211: MoveAlongJointSpline action?

jdlangs, 2019-08-19:
Hello,

The new action server to support executing splines directly is definitely great. It causes some issues for use though because we have planning software that operates in joint space and the current spline only accepts Cartesian input. Would you be willing to also have splines in joint space, via `SplineJP` in Sunrise?

Thanks,
Josh

jdlangs, 2019-08-19:
I should add we're in the process of implementing and test this now so there isn't any expectation of having to develop this yourselves.

SalvoVirga, 2019-08-19:
> Would you be willing to also have splines in joint space, via SplineJP in Sunrise?

Absolutely! But there is no need to ask!
This repo should be driven by community needs and efforts. If you have a use case / need for it, then implementing it yourself and giving it back to the community is the right approach!
Asking for a feature is the second choice if you have no means of implementing it, but due to the limited workforce it could take ages.