IFL-CAMP/iiwa_stack #214: Update Issue Ilegal StateException position violates limit

leitoxve, 2019-08-21:
Hello, **Thank** you for the effort you have made. 
I have done an update
Before It worked fine be the KONI port, I just wanted to try more services
I Update the stack both in my rosmachine and in the cabinet and i change to port X66
I **can connect the robot to ROS machine**, I can even **command the robot in joint position**
I  **can use moveit connected to my robot**
but when I want to command it by cartesian position the iiwa show me an error:
**java.lang.IlegalStateException:SSR CompoundRequest [0x000500020002C505]is invalid : Command Position violates limit A2 A6**
and I try to do it both by your rosclass-> ctpose.**setPose (Cart_Pose.poseStamped)** in a .cpp and by the terminal with:
**rostopic pub /iiwa/command/CartesianPose ometry_msgs/PoseStamped** "header:
  seq: 0
  stamp:
    secs: 0
    nsecs: 0
  frame_id: ''
pose:
  position:
    x: 0.2440
    y: -0.6440
    z: 0.80
  orientation:
    x: 0.0
    y: 1.0
    z: 0.0
    w: 0.0"
And i Have the same error.
and I have the axis limits like the wiki
I attached the photos for more details
**Thanks in advance**
[image]
SmartPadCapture
[image]

SalvoVirga, 2019-08-21:
Well, my first guess is that maybe that Cartesian pose you send might not be meaningful (especially the quaternion). 

Try this:
Set the robot in a "nice" configuration (in joint space, even with the SmartPad), read the flange current pose (as you did echoing `/iiwa/state/CartesianPose`) and send a Cartesian command to the robot changing using the current pose modified by some centimeters (2-5) in one direction only, keep the orientation the same the current one.

If the robot moves correctly, then the pose you are commanding the robot to right now is just not a meaningful one.

leitoxve, 2019-08-22:
Woow I'm surprised with the response speed, in fact something was not sending well, but is strange to me that my .cpp code is supposed to send something correct and it does not...
I'm going to check my code Thanks in advance and thanks for the great job.

SalvoVirga, 2019-08-22:
Be sure the pose your are sending is in the same reference frame that is set as your `EndpointFrame`. In the default case is "iiwa_link_0". You can chance that reference frame via a ROS service at runtime.

I'll close the issue, feel free to open it again if needed!

closed 2019-08-22 by SalvoVirga

Jorebs, 2019-12-26:
Hi and Merry Christmas! I also want to thank you for your hard work, it is extremely helpful!

I stumbled upon the exact same issue leitoxve found. I'm trying to command the robot to move using the rqt interface and using the /iiwa/command/CartesianPose and I get the following error message in the SmartPad:
**java.lang.IlegalStateException:SSR CompoundRequest [0x000500020002C505]is invalid : Command Position violates limit A2 A6**
I tried doing the same thing with  /iiwa/command/JointPosition and the motion is executed as expected. 

In order to solve the issue with the  /iiwa/command/CartesianPose, I tried what you suggested in your first reply (echoing for the current cartesian pose and then changing only the position along one of the axis, without changing the orientation). I obtain the exact same error message and no motion whatsoever. I'm quite certain that the pose I'm providing is a meaningful one, at least in the coordinate frame used by the echo /iiwa/state/CartesianPose. The message doesn't seem to change at all and the violated limits are always A2 and A6. I really don't know what else to try and if you had any suggestions I would be very thankful.

Could this be an issue related to the redundancy? There is no way (as far as I know) to provide the redundancy variables in the rqt interface. 
Could it be that the inputs I am providing through the rqt interface are in a different coordinate frame?

[image]

RuihanGao, 2021-05-18:
> Be sure the pose your are sending is in the same reference frame that is set as your `EndpointFrame`. In the default case is "iiwa_link_0". You can chance that reference frame via a ROS service at runtime.
> 
> EDIT: Actually, now there is a TF listener running on the Java side, so you should be able to send the command pose relative to any frame, forgot about that
> 
> I'll close the issue, feel free to open it again if needed!

Hi, thanks a lot for your reply. I encounter a similar issue. I use rosservice to set a new EndpointFrame, and I try to get the CartesianPose of that endpoint via rostopic `/iiwa/state/CartesianPose`. Is there any wat to do it? Thanks in advance!