IFL-CAMP/iiwa_stack #216: iiwa_stack not working with command/CartesianPose anymore

RoboticsKULeuven, 2019-08-22:
I'm trying to get my application working again with the updated iiwa_stack. (updated it on the Workbench and pulled the fresh repository)

I used to send PoseStamped() msgs to the /iiwa/command/CartesianPose topic and was able to move it until I ran all updates.

The Python code is straightforward and looks like this:

`
def cartPosCallback(cartPosData):
            self.cartPosition = cartPosData.poseStamped
self.cartPosPublisher = rospy.Publisher('/iiwa/command/CartesianPose', PoseStamped, queue_size=1)

self.cartPosSubscriber = rospy.Subscriber("iiwa/state/CartesianPose", CartesianPose, cartPosCallback, queue_size= QUEUE_SIZE)

def getCartPos(self):
        """
        returns the current joint position transformed to a cartesian pose using the iiwa's forward kinematic chain
        @return cartesian pose = a 4x4 matrix T = [R,p] with R 3x3, p = 3x1 and an extra line of [0,0,0,1]
        """
        cartPos = np.ones([4,4])
        pos_vec = KDL.Vector(self.cartPosition.pose.position.x,self.cartPosition.pose.position.y, self.cartPosition.pose.position.z)
        rot_m = KDL.Rotation.Quaternion(self.cartPosition.pose.orientation.x, self.cartPosition.pose.orientation.y, self.cartPosition.pose.orientation.z, self.cartPosition.pose.orientation.w)
        cartPos = tf.toMatrix(KDL.Frame(rot_m, pos_vec))
        return cartPos

def move_CartPos2(self, cartPos):
        """
        move the iiwa to the defined cartesian position 
        @param cartPos = 4x4 matrix containing a 3x3 Rotation matrix and 3x1 position vector with 
        last line [0,0,0,1]
        @param velocity = a joint velocitiy to be given to all 7 joints
        
        This function transforms the given cartesian position into a joint position for the iiwa robot
        and then calls the move_jntPos(jntPos) function to move the iiwa in place
        """
        # TODO: make it so that it sends real cartesian positions to the IIWA instead of joint positions.
        success = False
        try: 
            target = PoseStamped()
            target.pose.position.x = cartPos[0,3]
            target.pose.position.y = cartPos[1,3]
            target.pose.position.z = cartPos[2,3]
            quat = tf.fromMatrix(cartPos).M.GetQuaternion()
            target.pose.orientation.x = quat[0]
            target.pose.orientation.y = quat[1]
            target.pose.orientation.z = quat[2]
            target.pose.orientation.w = quat[3]
            
            
            #publish
            target.header.seq =1
            target.header.stamp = rospy.Time.now()
            target.header.frame_id = "world"
            self.targetCartPose = target
            self.cartPosPublisher.publish(target)
#           
            rospy.loginfo(target)
            self.cartPosPublisher.publish(target)
            self.rate.sleep()
            return True
                
        except rospy.ROSInterruptException:
            return success
`

Getting the CartesianPose is no problem, I've followed your update and that is fine but commanding the iiwa to move to another CartesianPose is not working. I've checked when running the code that it appears on the /command topic with 
`rostopic echo -c /iiwa/command/CartesianPose` and it does appear from the moment I send it. 
Might the problem be in the updated Sunrise Workbench application. that it is not receiving/processing the command?

This issue is related to #186 . 

With kind regards,
Zeno

SalvoVirga, 2019-08-22:
Hum, the code seems incomplete.
How do you call `move_CartPos2`? With which parameter?

You should try to read the current Cartesian pose and send it back to the robot changing it only in one direction by 2-5 cm, keeping the orientation the same as the current one.

Did you try to command in joint space? If that work but the Cartesian commands don't, it usually means that you are not sending meaningful poses.

RoboticsKULeuven, 2019-08-23:
Basically the program is a class `IIWA_Robot` so the way I use it to test if I can move the robot is as follows:

      rospy.init_node("iiwa_commander")
      iiwa = IIWA_Robot()
      destination = iiwa.getCartPos()
      destination[1,3] += 0.2
      iiwa.move_CartPos2

Which is already trying what you suggested, getting the current pose and changing it 2-5cm. But then again nothing happens. I also tried just doing it through a terminal, publishing it with ros topics as you can see in the pictures below:
First picture shows the echoed `/iiwa/state/CartesianPose` on the right and on the left a slightly altered PoseStamped msg to publish to the `iiwa/command/CartesianPose` topic.

The second picture shows the same on the left but on the right is an echo of the `iiwa/command/CartesianPose`, showing that me publishing is does put it on the topic.
(same happens when I did it through my Python program) 

[image]
[image]

So that leaves me wondering, if it shows up on the `iiwa/command/CartesianPose` topic, from there on `iiwa_ros_java` takes over on the Sunrise application so maybe that is not receiving it but I don't know where to start debugging that.

RoboticsKULeuven, 2019-08-23:
It appears that my `~/.bashrc` got compromised and the `ROS_IP` setting got removed, at least now I'm getting position errors again.

I didn't occur to me right away that those settings were removed because I was still able to get data out of the robot.

closed 2019-08-23 by RoboticsKULeuven