IFL-CAMP/iiwa_stack #224: missing iiwaRos helper class

tassos, 2019-09-17:
In the [documentation](https://github.com/IFL-CAMP/iiwa_stack/wiki/commandrobot) it is mentioned that a helper class `iiwaRos` is provided. However, it seems that the definition of `iiwaRos` in the __iiwa_ros__ package seems to have been removed [earlier this year](https://github.com/IFL-CAMP/iiwa_stack/commit/a7fb19ce1fde7dba1b73d981066043ffe8205453#diff-b30637227d4c006b6e63ae8620e1d1c4)

What is the recommended way to initialize an equivalent `iiwaRos` object? What class should be used?

tassos, 2019-09-18:
Sorry, it seems that the definition of `iiwaRos` was removed in 140fea62

SalvoVirga, 2019-09-18:
Yep, the documentation is still outdated, I'll try to bump it over some evenings...

Have a look at the headers in the [`iiwa_ros` package](https://github.com/IFL-CAMP/iiwa_stack/tree/master/iiwa_ros/include/iiwa_ros). Now that package provides classes for the individual uses (robot states, commands and clients to use the exposes services), their usage should be straightforward.

e.g. to read the robot Cartesian pose (written from the top of my head):

```cpp
#include <iiwa_ros/cartesian_pose.hpp>

int main() {

// Initialize ros first, ros::init, asyncSpinner etc
[...]

CartesianPose cartesian_pose;
cartesian_pose.init("iiwa");
[...]

auto current_pose = cartesian_pose.getPose();
}
```

You can also initialize those classes passing a callback that will be called every time a new message arrives.

tassos, 2019-09-18:
Ok I see, thanks a lot @SalvoVirga.

Now, the question is: do I need to initialize a different object for e.g. getting and changing the pose? Or is there an superclass object that has all the methods defined in the individual headers? If I understand the code correctly, the `Robot` class seems to be doing this, right?

So, could I do

```cpp
#include <iiwa_ros/iiwa_ros.hpp>

init main() {

  Robot my_robot;
  my_robot.init('iiwa');

  auto current_pose = my_robot.getPose();
  void my_robot.setPose(current_pose);
}
```

SalvoVirga, 2019-09-18:
`Robot` in `iiwa_ros.hpp` is an abstract class, you cannot use an object of that type.
You need to include and instantiate all the objects that you need (e.g. one for getting the cartesian pose state, one for commanding it). It was different before, but we changed to avoid to have every user creating a ton of publishers and subscribers even if they don't use them.

tassos, 2019-09-18:
Ok, I see. Thanks for the clarification!

closed 2019-09-18 by tassos