IFL-CAMP/iiwa_stack #226: How to start the action servers?

GuancongLuo, 2019-09-24:
HI!@SalvoVirga   @exo-core 

I want to test the spline_motion_demo. When I `rosrun iiwa_actions spline_motion_demo`, I get the `ROSINFO'Waiting for action servers to start'`. Should I need to write the servers code and start the servers code?
[https://github.com/exo-core/iiwa_stack_examples/tree/development](url)
Thank you so much.

exo-core, 2019-09-24:
Hi @GuancongLuo,

starting the ROSSmartServo application on the iiwa should be all you need. In case the spline_motion_demo is still waiting to connect you have probably a more general connection error. Make sure that all IP addresses are configured correctly and the you you have set the correct `ROS_MASTER_IP`.

Please note that the actions are only wrapping exiting functionality of KUKA Sunrise and thus are **not available in simulation** (we have been considering to build a similar interface wrapping MoveIt! for the simulation though).

GuancongLuo, 2019-09-30:
Thank you so much. As you said, it was working after I set the correct `ROS_MASTER_IP`.

closed 2019-09-30 by GuancongLuo