IFL-CAMP/iiwa_stack #227: trajectory following precision and module required for ros

xarthurx, 2019-09-24:
Hi, not sure if this is the right place to ask.

I borrowed an iiwa robot from another lab and intend to use it on a project that requires precise Cartesian trajectory following, controlled from ROS.
As the robot doesn't have a servoing module and it cost 4100Euros, I have two questions before deciding to purchase it.

1. The guy from the other lab told me that iiwa has a vibration issue and not good for trajectory following (reposition is good, though). Is it true?
2. From the wiki, servoing module is the only one needed, given the system is above 1.10 (we have 1.11). I guess this is also true?

I've never used a Kuka robot before, thank you for reading and sorry for the spam.

exo-core, 2019-09-30:
From my experience:

- If you have a ROS controller for the trajectory (e.g. you execute a motion via MoveIt!) the path looks rather bad; especially for Cartesian motions
- If you use a point-ot-point motion by giving only a target pose via the ROS action interfaces the trajectory looks way smoother. I did not do any detailed analysis on the precision but it was well enough for our sue cases so far.

Regarding the required Sunrise modules: You can find a picture of the modules you need to install in the [Wiki](https://github.com/IFL-CAMP/iiwa_stack/wiki/sunrise_project_setup). I think the SmartServo module is the only one that is not included in the default package.

xarthurx, 2019-09-30:
@exo-core  Thanks for the info.

So does this mean that iiwa is not very suitable for Cartesian motions from ROS, if not configured specifically? Then may I ask how did you execute Cartesian motions using iiwa?

Yeah, I contacted KUKA and our lab didn't buy that module... I may need to write sth for the sunrise workbench, still trying to find a solution.