IFL-CAMP/iiwa_stack #228: Load setting

leitoxve, 2019-09-25:
Hi, my iiwa has a tool that weighs about 2.7Kg and when I try to call a ForceDesired  it tells me:
`[ERROR] [1569421339.265578507]: /iiwa/configuration/ConfigureControlMode failed, Java error: java.lang.IllegalStateException: Load Mass Validation failed 
axisTauExtMsr[3] of LBRExtTorqueObs@LBR_iiwa_14_R820_1 is 10.6684813479544 should be less than 10
`
I can't change control mode either, and the iiwa/ Load Data /tool Mas is correct and selected in the KCP
I would like to know how I can make or adjust the Cartesian Wrench Force to  "0" or change tool mas settings  when the tool is plugged in robot to be able to call the functions of ConfigureControlMode ForceDesired.

exo-core, 2019-09-30:
You need to create an according tool in Sunrise and set the `/iiwa/toolName` rosparam to load it as described in the [Wiki](https://github.com/IFL-CAMP/iiwa_stack/wiki/attachtool). You can check the external torque on each axis on the SmartPad (make sure you select "Application Tool" in the payload settings).

GuancongLuo, 2019-10-09:
@leitoxve Did you solve this problem? Recently, I meet the same problem. Could you share how do you solve this problem?